ArduPilot/ardupilot #7902: Copter: drone not keeping yaw setpoint of SET_ATTITUDE_TARGET in GUIDED

deb0ch, 2018-03-12:
#### Issue details

We have been facing a problem for some time now and it looks a lot like a bug in the GUIDED mode.

We are controlling the drone’s attitude using the SET_ATTITUDE_TARGET message and sending quaternions, with the bitmask for the yaw set to absolute yaw angle (instead of yaw rate). The goal of that is to control the drone’s position around an object using a lidar.

The problem is that sometimes (often and randomly) the drone will rotate 20º to 60+º on its yaw axis, throwing off position control. We checked and double checked, the yaw we are sending in the quaternion is constant. It really is Ardupilot drifting away from its setpoint.

In the logs, the Yaw and DesYaw (that we believe is Ardupilot’s internal yaw setpoint) are totally desynchronised during the event.

In a later attempt, we replaced the bitfield for yaw with 0 in the SET_ATTITUDE_TARGET message, to have yaw rate control instead. **The exact same problem is happening with a yaw rate set to 0.**

This is from the code that we used to send the messages:

```python
def send_attitude_target(self, roll_angle=0.0, pitch_angle=0.0, yaw_angle=0.0, thrust=0.5):
    """Send an attitude command to the drone.

    At the time of writting, the drone's attitude must be commanded using a
    quaternion, so all angles must be commanded simultaneously.

    roll_angle, pitch_angle and yaw_angle are in radians.
    thrust is a number between 0 and 1, 0.5 having the special meaning
    of "maintain your current altitude".

    """
    msg = self.copter.message_factory.set_attitude_target_encode(
        0,
        0,                                         # target system
        0,                                         # target component
        0b00000100,                                # type mask: bit 1 is LSB
        self._to_quaternion(roll_angle, pitch_angle, yaw_angle),    # q
        0,                                         # body roll rate in radian
        0,                                         # body pitch rate in radian
        0,                                         # body yaw rate in radian
        thrust)                                    # thrust: < 0.5 descend, = 0.5 hold, > 0.5 ascend
    self.copter.send_mavlink(msg)
    self._log("send_attitude target: roll=%7.4f, pitch=%7.4f, yaw=%7.4f, thrust=%6.4f"
             % (roll_angle, pitch_angle, yaw_angle, thrust))

def _to_quaternion(self, roll=0.0, pitch=0.0, yaw=0.0):
    """Convert Euler angles in radians to quaternions."""
    t0 = math.cos(yaw * 0.5)
    t1 = math.sin(yaw * 0.5)
    t2 = math.cos(roll * 0.5)
    t3 = math.sin(roll * 0.5)
    t4 = math.cos(pitch * 0.5)
    t5 = math.sin(pitch * 0.5)
    w = t0 * t2 * t4 + t1 * t3 * t5
    x = t0 * t3 * t4 - t1 * t2 * t5
    y = t0 * t2 * t5 + t1 * t3 * t4
    z = t1 * t2 * t4 - t0 * t3 * t5
    return [w, x, y, z]
```

We previously had other yaw issues due to vibrations and bad props, that we fixed already: https://discuss.ardupilot.org/t/copter-roll-and-pitch-affecting-yaw-in-stabilize/26076

We checked all the mechanical structure, it is OK without too much vibrations. The PID settings are from a successful Copter 3.5 POSHOLD Autotune.

In the other modes (STABILIZE, ALT_HOLD, LOITER), the drone flies flawlessly with perfect response to all inputs.

#### Version
ArduCopter 3.5

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Octa-quad of about 6 kilograms. SET_ATTITUDE_TARGET messages sent from Dronekit on a Raspberry Pi through serial link.

#### Hardware type
PixHawk

#### Logs

Logs of a short flight where the bug was reproduced: https://drive.google.com/open?id=0B7L1T2gKDuFocHlacWxyRHdHdkp5aG51R1JfN093UlItbmxF

Video from a different flight demonstrating the issue: https://drive.google.com/open?id=1vBfKHn5ACHXV7rt8Eos3q7RvsH_1wG45

On the video you can see the drone controlling its roll and pitch position around the pillar, in GUIDED mode without any pilot input except for thrust.
Then you can see rotate suddenly (clockwise seen from above) and drift to the left. The drifting is normal but unwanted, being the drone trying to keep the pillar in front of it and in range.
**This yaw axis rotation is not wanted nor ordered by our sowftware, but by Ardupilot itself internally, and throwing off our position control as you can see on he video**.

rmackay9, 2018-03-22:
This is not really the place for support issues but in any case, I'm pretty sure this is a tuning issue.  The DesYaw and Yaw deviating backs up this theory.  The logs show overly active throttle and motor outputs likely caused by gains that are too high.  This is leading to saturation of the motors which is probably leading to the loss of yaw control as it needs to make sacrifices and the yaw axis is often the first to be sacrificed.

Looking at the logs these parameter changes may help:

- ACCEL_Z_P and ACCEL_Z_I are high (0.75 and 1.5).
- ATC_RAT_PIT_P/I/D are far too high perhaps this is the result ofa  bad autotune?.  I'd return these to the defaults which will very likely be better.  ATC_RAT_PIT_P/I and ATC_RAT_RLL_P/I to 0.135.  ATC_RAT_PIT_D/ATC_RAT_RLL_D to 0.0036.

[image]
[image]

closed 2018-03-22 by rmackay9

deb0ch, 2018-03-22:
Thank you so much Randy for your insight ! We did previously post on [discuss.ardupilot.org](discuss.ardupilot.org) without success, so tried here because it seemed to us that it might be an actual bug.

We were totally stuck on this without any clue of how to fix it or what went wrong. Your help here is precious and life-saving ! We wouldn't have been able to tell if the throttle and motor outputs were overly active or too high, without experience seeing these signals in normal conditions.

What you said seems to make a lot of sense, we will have an opportunity to test that this week end.

Though, the PID settings that we came up with were from an autotune in POS_HOLD that went smoothly, right after a clean install and starting from the default settings. If these are the right conditions to apply an autotune, doesn't our issue mean that the autotune algorithm needs some refining ?

rmackay9, 2018-03-23:
@deb0ch, glad to help.

The AutoTune sometimes doesn't work but I think the most common cause is something moving around on the frame (i.e. a gimbal) or environmental conditions (i.e. too much wind).  The algorithm looks for little dips in the rotation rate but it can be thrown off by other causes of changes in rotation rate.

It's possible that more could be done to identify frames that can't be autotuned.. I'll think about it and perhaps talk with Leonard about it.  It's tricky though..

deb0ch, 2018-03-23:
We *do* have a gimbal, maybe that is what threw off our autotune ?

I get that such calculation is tricky and a perfect solution, if it exists, might take a long time to reach. In the meantime, I never suspected until you mentioned it now that our gimbal could have thrown off the autotune.

A very quick, simple and easy way of avoiding such problems would be **a big, visible warning against the use of gimbals or other causes of disturbance during the autotune, with guidelines of how to setup the drone** for the autotune ? I would see such a warning useful both in the ground stations and in the documentation (the official tutorial page for autotuning that we followed).

For instance, should we have unplugged and immobilized our gimbal, or simply take it off the drone ? I don't know which approach would have had the least impact on the autotune, as since the gimbal is heavy and all the way at the front it has a big impact on the drone's dynamics.