ArduPilot/ardupilot #7913: UAVCAN for SITL

khancyr, 2018-03-13:
#### Issue details
UAVCAN is available on linux and working wuth socketcan. This allows to simulate CANbus over socket on a computer. 
Then it would be interesting to get an inclusion of some UAVCAN nodes into SITL. 

This should modify sitl_state and register uavcan as new external device on sitl model. This will allow to update SITL with simulated CAN and provide feedback to controler.

auturgy, 2018-07-30:
Can this be closed?

khancyr, 2018-07-30:
@auturgy It has been done yet !

auturgy, 2018-07-30:
Sorry. Just found it. https://github.com/ArduPilot/ardupilot/pull/8509 hasn’t been merged yet.

Regards,

James

IamPete1, 2021-07-26:
We have SITL CAN now with https://github.com/ArduPilot/ardupilot/pull/14477

closed 2021-07-26 by IamPete1