ArduPilot/ardupilot #7914: Add MAVFTP

proficnc, 2018-03-13:
#### Issue details
Add MAVFTP to allow simple file transfer for peripheral firmware updates, and drag and drop log transfer 

#### Version
Master

#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
All

#### Hardware type
STM32 with SD card
#### Logs
NA

vooon, 2018-03-13:
Actually i'm not sure that full implementation of PX4 FTP is needed. Mounting FCU's filesystem via fuse is funny but not very practical (and almost unusable since context limited to one file).

Needed operations:
1. List files
2. Download file - request FW to stream file chunks. No ACK, or other messages from GCS.
    At completion GCS may request missing chunks.
3. Upload file - same but in opposite direction.
4. mkdir/rm

anandbaskaran, 2019-03-05:
Thanks got posting this. I am looking for the same feature. 

However, is there any way currently to get the drone logs on the custom GC?

Kiwa21, 2019-11-28:
Any news on this subject ?

WickedShell, 2019-11-30:
@Kiwa21 support was added with #12533. QGC needs a couple of patches to understand that ArduPilot supports it now (and to adjust unrealistic timeouts), but it works in Mission Planner, a very small patch for QGC (although current QGC builds hide the entire FTP interface anyways), and in MAVProxy.

closed 2019-11-30 by WickedShell

liang-tang, 2019-12-09:
how to test this?

peterbarker, 2019-12-09:
On Sun, 8 Dec 2019, liang.tang wrote:

> how to test this?

connect with a daily build of qgroundcontrol

liang-tang, 2019-12-09:
thanks for your help, i'll try it later.