ArduPilot/ardupilot #7927: QuadPlane disarms VTOL motors in flight

expntly, 2018-03-14:
#### Issue details
In an AUTO mission right when the vehicle de-transitioned from fixed-wing to VTOL (X8 quadplane), one of the motors (C6) stopped spinning, later confirmed on the ground with a cold motor and seeing the motor beep before disarming).  At this point it started yawing aggressively clockwise, and we had to take over manually in QLOITER to prevent a crash.  This didn't quite stop the yaw and we descended pretty much in monocopter mode all the way down.  Obviously very hard/scary to land.

We had seen motors/ESC disarm randomly on the ground when doing quick checks (rapid throttle up and down) before, but never in flight.  Looking at the logs, that C6 motor in green is asked to work hard.  However what's even more surprising is C11 in pink, which is dead flat at 1150PWM.

[image]

#### Version
3.8.2

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Octaquadplane

#### Hardware type
PH2.1

#### Logs
DF log (170MB):  https://drive.google.com/open?id=1SuhECreVJjpuR45iRPNMPrGDVwqU1mrs

auturgy, 2018-03-14:
Looks like C6 lost sync or failed, and C11 was commanded low by the autopilot to try to control yaw. This isn’t ‘disarming’ by ArduPilot - it’s most likely a problem with your setup.

Regards,

James

lthall, 2018-03-15:
The code is operating correctly for a lost motor. There is no sign that this is a code issue.

Closing

closed 2018-03-15 by lthall

charlesacknin, 2018-03-15:
Thanks for the quick replies!

@lthall  We've been flying extensively with this octaquadplane config for the last year.  We've "lost motors" more than half a dozen times (burned ESC, prop strike, motor fire, mount failure), and we've never had a vehicle enter this spiraling uncontrollable yaw.  Heck we've even had [this](https://drive.google.com/file/d/10WSk7EO0CPKful3RawHgkUwtRkdByjN5/view?usp=sharing) and the vehicle landed normally as if nothing happened -- yes the lower prop is entirely gone and the upper one has more than 30% missing.

@auturgy The moment it landed, we re-armed and did a short QSTABILIZE/QLOITER flight (it's in the same log so you can look at it) without problem.  So the hardware seems fine.

expntly, 2018-03-15:
Thanks for the quick replies!

@lthall We've been flying extensively with this octaquadplane config for the last year. We've "lost motors" more than half a dozen times (burned ESC, prop strike, motor fire, mount failure), and we've never had a vehicle enter this spiraling uncontrollable yaw. Heck we've even had this and the vehicle landed normally as if nothing happened -- yes the lower prop is entirely gone and the upper one has more than 30% missing.

@auturgy The moment it landed, we re-armed and did a short QSTABILIZE/QLOITER flight (it's in the same log so you can look at it) without problem. So the hardware seems fine.

lthall, 2018-03-16:
You have observed sync loss on the ground. Your logs clearly show motor loss and you confirmed that with temperature measurements on the ground.

Being able to take off again after also supports sync loss.

I would not discribe this as showing your hardware is fine.

Are you suggesting that I was unreasonable to close this issue?

expntly, 2018-03-17:
@lthall You may be right that the hardware is the root cause.  However I'm pointing out the fact that we have motor failures all the time and we've never seen the software react that way.  Does that make sense?  If you can explain why RCOU.C11 is stuck at 1150PWM (see graph in initial post), that might help.  Thanks

lthall, 2018-03-18:
@expntly To balance the roll and pitch of the vehicle when a motor is lost the opposite motor or motors must drop in throttle. So you will see the lost motors go high and the physically opposite motors go low to keep the thrust balanced. This is what keeps the aircraft from flipping over and tumbling to the ground. It is normal in these conditions to lose yaw control but it depends on the CG of the airframe and the motors lost.

If you would like to discuss this further I would suggest opening a discussion on the discussion forums and pinging me there. I will be happy to talk further over there.

charlesacknin, 2018-03-18:
@lthall Sure: https://discuss.ardupilot.org/t/loss-of-control-after-motor-failure-on-octaquadplane/26997

expntly, 2018-03-18:
@lthall Sure: https://discuss.ardupilot.org/t/loss-of-control-after-motor-failure-on-octaquadplane/26997