ArduPilot/ardupilot #7931: Eph and Epv values in Fake GPS through mavros are set to 9999

mellonUAV, 2018-03-15:
Hello,

I am using mavros fake_gps plugins to send GPS data to ArduPilot. I can send lat and long values pretty well. The problem I have is Ardupilot doesn't respond to the eph and epv values that I send and just converts them to 9999. How can I fix this issue? Please help!

```
# fake_gps
fake_gps:
  # select data source
  use_mocap: true         # ~mocap/pose
  mocap_transform: false   # ~mocap/tf instead of pose
  use_vision: false       # ~vision (pose)
  # origin (default: Zürich)
  geo_origin:
    lat: 39.894470          # latitude [degrees] 47.3667
    lon: 32.777973           # longitude [degrees] 8.5500
    alt: 938.0            # altitude (height over the WGS-84 ellipsoid) [meters]
  eph: 0.5
  epv: 0.5
  satellites_visible: 12   # virtual number of visible satellites
  fix_type: 3             # type of GPS fix (default: 3D)
  tf:
    listen: false
    send: false           # send TF?
    frame_id: "map"       # TF frame_id
    child_frame_id: "fix" # TF child_frame_id
    rate_limit: 10.0      # TF rate
  gps_rate: 5.0           # GPS data publishing rate
```
[image]

Best regards,
Melih

#### Version
APM 3.5.5

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
quadrotor

#### Hardware type
Erle Brain 3

JustFineD, 2018-03-15:
Hi @mellonUAV ,
I also had this problem. 
As you can see in AP_GPS_MAV.cpp (3.5.5) there is a bug in the code:
state.hdop = MAX(packet.eph, 9999);
state.vdop = MAX(packet.epv, 9999);

You only need to change these lines and rebuild the project.
The problem is already solved in Copter-3.6

WickedShell, 2018-03-28:
As noted is correct in master.

closed 2018-03-28 by WickedShell