ArduPilot/ardupilot #7932: Plane::send_nav_controller_output uses incorrect airspeed units

lekston, 2018-03-15:
#### Issue details
Plane::send_nav_controller_output reports airspeed error in [cm] whereas mavlink specifies it in [m].
A (very brief) search in Mission Planner, qGroundControl and Droid Planner/Tower they do not use this message so I believe there should be no problems with correcting it.

#### Version
3.8.4

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

Naterater, 2018-03-15:
I'd like airspeed in general to be in m/s with 2 decimal places instead of cm/s. That might require some scaling elsewhere though.

As a reference, about 15 parameters use cm/s (mostly quadplane parameters).
The other ~50 parameters use meters.

Takeoff and landing speeds are in m/s, wind speeds are in m/s, EKF values are in m/s. I would think that cruise, ground, and air speed should be the same way.

magicrub, 2018-03-15:
I agree that m/s is easier than cm/s. They were put in that way for a good reason: floating point math is expensive on 8bit cpus. However, that time has passed and now the problem is param names are hard to change for compatibility reasons.

However, what @lekston is saying is we are not obeying the MAVLink spec and transmitting the data in wrong units. That is something that needs to get fixed.

magicrub, 2018-03-15:
pending PR: https://github.com/ArduPilot/ardupilot/pull/7933

Naterater, 2018-03-15:
Does it also need to be corrected in Mission Planner?

magicrub, 2018-03-15:
>A (very brief) search in Mission Planner, qGroundControl and Droid Planner/Tower they do not use this message

needs to be confirmed before that PR gets merged

lekston, 2018-03-16:
I did a brief search in their repositories and I found no "mavlink_nav_controller_" nor "mavlink_msg_nav_controller" in there.

meee1, 2018-03-20:
Missionplanner does infact use it
https://github.com/ArduPilot/MissionPlanner/blob/master/ExtLibs/ArduPilot/CurrentState.cs#L2316

lekston, 2018-03-20:
My apologies for overlooking this. Thanks Michael! I will know where to look next time.

amilcarlucas, 2021-09-22:
ping?