ArduPilot/ardupilot #7934: Plane: incorrect loiter heading action - early exit

lekston, 2018-03-15:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
When flying to a distant loter_to_alt waypoint, the altitude target is often reached before the airplane established in a loiter pattern (i.e. still on L1 capture phase). In this case the heading verification will be incorrect due to an ill behaved back-up condition. On SITL and in real Plane I typically see an early exit: ~90deg early because of the backup condition of verifying (i.e completing) the loiter_to_alt command after passing a (loiter.sum_cd > loiter.total_cd), where loiter.total_cd is incorrectly initialized because we in the above described scenario we are still in the capture phase (that is heading towards the center of loiter), and the loiter.total_cd target is off by about 90deg. (see Plane::verify_loiter_heading(bool) in commangs_logic.cpp.). This can be solved by using an alternative backup condition: instead of counting the total_cd, we might extend the acceptance criteria after every loiter. That is during first loiter we use +/110deg of margin in the next +/-20deg and so on. Please see PR referred below.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

magicrub, 2018-03-15:
can you post a screen shot showing the problem?

lekston, 2018-03-16:
[image]

Above an example of early exit: entering loiter_to_alt in level flight. Below, same flight with fix from #7935 

[image]

magicrub, 2018-03-16:
Won't that break tangent exits?

lekston, 2018-03-16:
I will check that, but as you can see on the first screenshot the current firmware does not guarantee tangent exit neither.

lekston, 2018-03-20:
@magicrub So I had a look at this again, and in fact the suggested fix will have at least as good behaviour for tangent exits as the original with the benefit of avoiding the bug presented in screents above. Just to be more specific: on the 2nd screen the loiter was verified (completed) at a point of target exit (from the south-southe-west part of the loiter toward the wpt # 6), however since I did set all loiter parameters to zero, hence the next_WP_loc.flags.loiter_xtrack flag was also zero so the airplane intercepted the original leg from the center of loiter to wpt # 6. I will run the same sim scenario in a moment (just need 15mins) with tangent exit enabled to demonstrate.

lekston, 2018-03-20:
Here is an example showing that tangent exit baheaviour is not changed by the proposed fix #7935 
[image]

magicrub, 2018-03-20:
I'll experiment with a few loiter missions and if all goes well I'll merge.

lekston, 2018-03-20:
Great, let me know if there is anything I can help with.

magicrub, 2018-03-20:
merged your PR https://github.com/ArduPilot/ardupilot/pull/7935, thank you for tracking this down!

closed 2018-03-20 by magicrub