ArduPilot/ardupilot #7941: libraries/GCS_Mavlink/include/mavlink/v2.0/mavlink_helpers.h not found

aniskoubaa, 2018-03-16:
#### Issue details
The folder 
libraries/GCS_Mavlink/include/mavlink/v2.0/
is no longer available

The code cannot compile without it

#### Version
Latest version on https://github.com/ArduPilot/ardupilot/ as of today.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

khancyr, 2018-03-16:
Hello,

This folder is generated on compilation time from the mavlink submodule

magicrub, 2018-03-16:
can you show your compile output? This gets generated at compile time so it's probably failing in there somewhere but somehow still trying to continue and failing elsewhere

aniskoubaa, 2018-03-16:
Actually it seems something is going wrong.
I am using a VM machine with Ubuntu Kinetic. 

I followed these steps:
`git clone https://github.com/ArduPilot/ardupilot.git`
Run the install-prereqs-ubuntu.sh script by running this command
`./ardupilot/Tools/scripts/install-prereqs-ubuntu.sh`

Then I installed the following packages

```
cd
 
sudo add-apt-repository ppa:george-edison55/cmake-3.x -y
sudo apt-get update
sudo apt-get install cmake
 
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-get update
sudo apt-get install g++-4.9
 
sudo add-apt-repository ppa:terry.guo/gcc-arm-embedded
sudo apt-get update
sudo apt-get install gcc-arm-none-eabi=4.9.3.2015q3-1trusty1
 
sudo apt-get update
sudo apt-get upgrade
```

I had an issue with
`sudo apt-get install gcc-arm-none-eabi=4.9.3.2015q3-1trusty1`

so I run the following instead
`sudo apt-get install gcc-arm-none-eabi`
without version and it seems to work.

Then I run
`./waf list_boards`
then:
`./waf configure --board px4-v2`

Here I had an issue, I got the following error

```
riotu@ubuntu:~/ardupilot$ ./waf configure --board px4-v2
Setting top to                           : /home/riotu/ardupilot 
Setting out to                           : /home/riotu/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : px4-v2 
Checking for program 'arm-none-eabi-ar'  : /usr/bin/arm-none-eabi-ar 
Using toolchain                          : arm-none-eabi 
Checking for 'g++' (C++ compiler)        : /usr/bin/arm-none-eabi-g++ 
Checking for 'gcc' (C compiler)          : /usr/bin/arm-none-eabi-gcc 
Traceback (most recent call last):
  File "/home/riotu/ardupilot/modules/waf/waflib/Scripting.py", line 165, in waf_entry_point
    run_commands()
  File "/home/riotu/ardupilot/modules/waf/waflib/Scripting.py", line 266, in run_commands
    ctx = run_command(cmd_name)
  File "/home/riotu/ardupilot/modules/waf/waflib/Scripting.py", line 250, in run_command
    ctx.execute()
  File "/home/riotu/ardupilot/modules/waf/waflib/Configure.py", line 159, in execute
    super(ConfigurationContext, self).execute()
  File "/home/riotu/ardupilot/modules/waf/waflib/Context.py", line 205, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/home/riotu/ardupilot/modules/waf/waflib/Context.py", line 287, in recurse
    user_function(self)
  File "/home/riotu/ardupilot/wscript", line 168, in configure
    cfg.get_board().configure(cfg)
  File "Tools/ardupilotwaf/boards.py", line 669, in configure
    super(px4, self).configure(cfg)
[... 59 lines cut ...]

Then I compiled as:
```
riotu@ubuntu:~/ardupilot$ ./waf copter
Waf: Entering directory `/home/riotu/ardupilot/build/sitl'
Waf: Leaving directory `/home/riotu/ardupilot/build/sitl'

BUILD SUMMARY
Build directory: /home/riotu/ardupilot/build/sitl
Target               Text     Data  BSS    Total  
--------------------------------------------------
bin/arducopter       1655334  1962  42464  1699760
bin/arducopter-heli  1613286  1962  42304  1657552

Build commands will be stored in build/sitl/compile_commands.json
'copter' finished successfully (2.004s)
```
and it showed a success

but in the folder libraries/GCS_Mavlink, there are no folders created after compilation as it would be expected.

I do not know if I missed any step or there is something wrong to make it work
Any help will be appreciated.

Thanks

khancyr, 2018-03-16:
We try to don't do support on the issue list, there is nothing wrong here. You will find the mavlink folder in build/sitl/libraries/GCS_MAVLink/include . I closing now, please open a topic on if you need more help : http://discuss.ardupilot.org/
Other points : 
gcc-arm-none-eabi isn't the same as our toolchain.
on px4-v2 error, try git submodule update --init --recursive
on navio2, you don't have arm-linux-gnueabihf compiler

closed 2018-03-16 by khancyr