ArduPilot/ardupilot #7948: Plane, feature request, RTL Steps or min height

SlipstreamFPV, 2018-03-17:
Setting ALT_HOLD_RTL to -1 makes the plane return at the altitude at the time of the failsafe which is often advantageous. But what if the failsafe occurs while the plane is very close to the ground, then we should be using a normal positive setting which is an issue if we have climbed to clear a mountain etc.

Is it possible to make an RTL mode where ALT_HOLD_RTL is the minimum height at which the plane will return however if the plane is higher it will also maintain that altitude?

Or even better “stepped” RTL, where there are two or three programmable steps. EG <1000m distance from home then RTL height = 100m; >1000m distance from home then RTL height = 300m

magicrub, 2018-03-17:
Interesting idea. 

Regarding the RTL with mountain in the way, have you ever tried using the terrain follow feature? It does that.

JeyPi, 2018-06-17:
My today's FPV experience:
I took off from a spot 280 meters AMSL (altitude above mean sea level)
I was remotely landing the plane while in the stabilized mode at an RC field that was a 1000 meters away from me and a bit lower - 240 AMSL.
I was gliding in for landing. All was great. Just before touch down, the plane lost RC link and the RTL mode kicked in. I was no longer in control. The ardupilot software took a right turn towards the Take off position without any climb. Just a turn. That of course was followed by a crush as the plane was so low and the tip of the wing touched the ground. My RTL altitude was set to 100 meters.
I suggest the code should be adjusted to climb the plane to a safe altitude before turning to the home position. I understand this could not suite to all platforms so at least this behavior could be user pick able.
The only data that survived was the camera footage. The flight deck data card was destroyed.

[Video of the event here](https://drive.google.com/open?id=1M_ZMg-GumfTnKhr5usVDDcZS5AaffCR2)

As you can see, the only thing needed was a little climb before the turn and plane would had been saved.

peterbarker, 2024-07-31:
We have RTL_CLIMB_MIN now - probably would have saved @JeyPi 's vehicle.

closed 2024-07-31 by peterbarker