ArduPilot/ardupilot #7958: Alt doesn't match BAlt or GPSAlt possibly causing uncommanded rise in altitude

optix2000, 2018-03-19:
#### Issue details
More of a question for how altitude is calculated.

I've had this happen a few times where while in Loiter, where the drone will shoot up into the sky without warning until it's brought into stabilize. From what I can gather in the logs, this comes from the fact that the FC thinks it's dropping altitude (`CTUN.Alt` starts plummeting), and thus it tries to climb (`DAlt` > `Alt`). However both `GPS.Alt` and `CTUN.BAlt` appear correct (and show the climb), and GPS has 12 Sats and <1 HDOP.
[image]

From what I can gather, this seems to happen when there's significant accelerometer clipping (known issue, will fix soon).

The question here is, why would the accelerometer override `CTUN.Alt`, when both `BAlt` and `GPS.Alt` should be considered good sensor values?
From my naive understanding of flight controllers, shouldn't altitude be calculated from all sensors, with a priority of the most reliable ones? ie Baro > GPS > Dead reckoning from accelerometers? This is of course assuming there's no pressure fluctuations from the barometer due to prop wash, and a good enough HDOP and 3D fix.

#### Version
Copter 3.5.5

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quad X

#### Hardware type
680mm frame
15x5 in props
Ublox M8N
Pixracer

#### Logs
Logs: [18-03-01_15-47-58.bin.zip](https://github.com/ArduPilot/ardupilot/files/1823852/18-03-01_15-47-58.bin.zip)

lthall, 2018-03-19:
Hi Optix,
This is a support question rather than an issue. Please post questions like this on the discussion forums.

To answer your question briefly the kalman filter takes gps alt, baro alt and accelerometers to form a best guess of the height. At the most basic level the accelerometers give the fast guess and the gps and baro inform what we should get over a longer time frame. So when the accelerometers say we are climbing quickly (because we are clipping) we get a sudden error increase. We are trying to ignore the accelerometers when we see clipping but this is a complex problem.

closed 2018-03-19 by lthall

optix2000, 2018-03-19:
Hey @lthall,

> This is a support question rather than an issue. Please post questions like this on the discussion forums.

I had actually brought this up in the discussion forums, but didn't get a solid answer, which is why I brought it up here. This _could_ have been a bug, but wanted some clarification before tagging it as such.

> We are trying to ignore the accelerometers when we see clipping but this is a complex problem.

This sort of reads as, "we don't ignore accelerometers yet when they're clipping, because it's a complex problem." Does this imply that there's currently no solution to these sorts of issues aside from soft mounting the FC? If so, are there any plans to (ie existing issues tracking it)?

From the discussion forums, this seems to be not an uncommon problem (vertical flyaways), and there doesn't seem to be any hard and fast solutions looking at previous discussions, or in the docs. There's many opportunities for this class of issues to be addressed, either in code or in documentation. I just want to make sure this is intended behavior (as un-ideal as it might be). If it's something that can be resolved in code, this can track the discussion for the fix, and if it can't be resolved in code, this can track the documentation for any hardware solutions or workarounds.

In this vein, so we can exhaust is there some parameter that can be tweaked to either clamp the error amount the accelerometer's can introduce, as an alternative to finding the perfect softmount material.