ArduPilot/ardupilot #7963: PLANE, maybe others: LOG_DISARMED not effective after first arm

Naterater, 2018-03-19:
#### Issue details
I have LOG_DISARMED set to 0. I disarm and re-arm after landing as I want to start a new log, but I never get new logs.

#### Version
Plane 3.8.4

#### Platform
[?] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Traditional Twin Electric Plane

#### Hardware type
Pixhawk 2.1

#### Logs

Be advised, this is a large 3-flight log with disarming in-between takeoffs.
https://1drv.ms/u/s!AunBK4rE1ZDC8htDzMUg5OPaF7qg

magicrub, 2018-03-19:
I believe it stops logging but then resumes when you re-arm. Same log file. Please confirm.

I don't think closing and starting new logfile is supported. Would be nice though.

Naterater, 2018-03-19:
I guess that's my bad on the interpretation of logging. It may be working as intended.

Naterater, 2018-03-19:
This makes sense now: LOG_FILE_DSRMROT closes the log and opens a new one.

closed 2018-03-19 by Naterater