ArduPilot/ardupilot #7972: Plane (Tail Sitter): abnormal motors shutdown during transition from "FBWA" to "Q_Stabilize"

robustini, 2018-03-20:
#### Issue details
For over two seconds my Tail Sitter's motors went off during a "FBWA" to "Q_Stabilize" transition.
We know that the artificial horizon repositioned during this transition, and I believe that the code did not correctly interpret the position.
Unfortunately for a MicroSD problem I had to disable the dataflash log, so only the telemetry log is available, but the problem is obvious.
As can be seen from the graph the two motors go to pwm 1000us, so switched off.

[image]

#### Version
ArduPlane V3.8.5beta1 (bd9fe3f0)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
VTOL Tail Sitter "non vectored"

#### Hardware type
Pixhawk, M8N GPS, without airspeed

#### Logs
Is in the video description:
https://drive.google.com/file/d/198hF2BoOlBNcgvJO8DJh4MCWXwIIV-5y/view

peterbarker, 2018-03-20:
This alone is interesting.  The mode seems to change instantly based on the channel switch - except for the last, bad, transition:

`graph RC_CHANNELS_RAW.chan8_raw`

bzucak, 2018-05-02:
I have the same problem, with my tailsitter during transition from FBWA to QHover. This power drop makes transition really unstable.

closed 2020-12-16 by robustini