ArduPilot/ardupilot #7976: Copter: Compassmot polls the battery excessively fast

WickedShell, 2018-03-21:
#### Issue details
Copter manually polls the battery read function at 50Hz within the compassmot, this in turn updates the resistance at 50Hz. The filter for this was setup and tuned at a 10Hz call rate, and thus will create different results then would be seen in normal usage.

This should probably be rate limited within the compassmot test, however I didn't want to address this until after the battery failsafe stuff is merged first due to the ever ongoing scope creep.

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

amilcarlucas, 2018-04-07:
Are all your battery PRs now in master ?

WickedShell, 2018-04-07:
Except #8025 which is stuck on me finishing moving some params I missed earlier.