ArduPilot/ardupilot #7978: Rover: move methods in commands_logic.cpp to mode_auto

rmackay9, 2018-03-21:
This is quite similar to [this Copter issue](https://github.com/ArduPilot/ardupilot/issues/7707).

We've restructured the Rover so that all the modes are classes (see mode.h, mode_auto.cpp, etc).  As a follow-up we should move the contents of the [commands_logic.cpp](https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/commands_logic.cpp) file into [mode_auto.cpp](https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/mode_auto.cpp).

This will include:

- removing the method definitions from [Rover.h](https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/Rover.h#L404) into [mode.h](https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/mode.h#L185).
- move the implementations from [commands_logic.cpp](https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/commands_logic.cpp) to [mode_auto.cpp](https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/mode_auto.cpp) and change the class the methods belong to so instead of starting with "Rover::" they start with "ModeAuto::".
- ensure the order that the methods appear in mode.h matches the order within mode_auto.cpp

bCamba, 2018-03-22:
@rmackay9 I have a question about the third bullet point. Should I change the methods order in mode.h file to match mode_auto.cpp or the other way around?

I worked on the Copter related issue #7707 so I can work on this one as well

naveentrtumkur, 2018-03-23:
@bCamba I would like to take up this issue. I'm getting familiar with arduPilot codebase and SITL testing. Solving bugs like this would be a good start for me. If you haven't started working on this, may I take up this issue?

rmackay9, 2018-03-24:
@bCamba, I think for the most part, the .h file has a good order so making the .cpp match the .h is best.

@naveentrtumkur, we have lots and lots of issues so let's see if we can find you another one :-).

bCamba, 2018-03-24:
@naveentrtumkur, I have not started yet. Feel free to tackle this one :)

naveentrtumkur, 2018-03-27:
@rmackay9 As part of this issue re-working, I was studying the different modes available. I didn't clearly understand loiter mode. It says "automatic horizontal acceleration with automatic throttle". My understanding was : In loiter mode, there would be automatic horizontal acceleration of vehicle with pauses after specific duration. Can you breifly explain loiter?

sarthak268, 2018-04-30:
Is this still open? If yes, please assign

peterbarker, 2018-04-30:
@cindyywang do you have a PR for this issue?

sarthak268, 2018-05-02:
Please assign

peterbarker, 2018-05-02:
@sarthak268 I've assigned this to me on the assumption you'll poke it.

sarthak268, 2018-05-04:
What all still needs to be done in this issue?

MohamedAliRashad, 2019-03-21:
#10847 I think this PR close the issue.

OXINARF, 2019-03-22:
This was done by #8019/#8023/#10827.

closed 2019-03-22 by OXINARF