ArduPilot/ardupilot #7982: runsim

yahyaelabiad, 2018-03-21:
Hi,

I am trying to run the SITL simulation and I have been following the tutorials till this step:

>./runsim.py xxx

it shows error messages:

Traceback (most recent call last):
  File "./runsim.py", line 8, in <module>
    import fdpexpect
ImportError: No module named fdpexpect

khancyr, 2018-03-21:
hello,

What tutorial are you following ? 
About the error, just do : 
````
sudo pip2 install -U pexpect
````

yahyaelabiad, 2018-03-21:
i'm following this tutorial http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html
but Actually i'm trying to launch jsbsim with arduplane  with this sim_vihecle -f jsbsim:Rascal
and making connection with filghtgear who's in different network
but it doesnt work

ps: it still give same error