ArduPilot/ardupilot #7983: HAL_BOARD_STORAGE_DIRECTORY needs to be run-time configurable

vilvo, 2018-03-21:
#### Issue details
When running ardupilot on Ubuntu Core, the [build-time defined HAL_BOARD_STORAGE_DIRECTORY](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/board/linux.h#L435) maps to non-writable location. 

Ubuntu Core defines snap-container writable areas through environment variables - https://docs.snapcraft.io/reference/env - this kind of application specific data would go under $SNAP_DATA

Proposed solution: make ardupilot system writable locations run-time configurable - e.g. parameter or environment variable.

#### Version
Development build on Linux from master-branch

#### Platform
[X] All

#### Hardware type
Linux

#### Logs
```
root@localhost:/snap/ardupilot/x3/bin# ./arduplane 
Failed to create /var/lib/ardupilot/ArduPlane.stg
```

lucasdemarchi, 2018-03-21:
You can define it to a writable location for your specific board. On Bebop/disco there's exactly this problem and the solution is to write elsewhere. See https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/board/linux.h#L40

Alternatively.... it's already runtime and if you pass `--log-directory /tmp` it should save the logs there, or anywhere else you want and can write to.

vilvo, 2018-03-22:
Thanks @lucasdemarchi. Yes, build time configuration will work of course.

I tested your alternative runtime proposal but it did not work:
```root@localhost:/snap/ardupilot/current# bin/arduplane --help
Usage: -A uartAPath -B uartBPath -C uartCPath -D uartDPath -E uartEPath -F uartFPath
Options:
        serial:
                    -A /dev/ttyO4
                          -B /dev/ttyS1
        networking tcp:   
                          -C tcp:192.168.2.15:1243:wait
                          -A tcp:11.0.0.2:5678
                          -A udp:11.0.0.2:14550
        networking UDP:   
                          -A udp:11.0.0.255:14550:bcast
                          -A udpin:0.0.0.0:14550
        custom log path:  
                          --log-directory /var/APM/logs
                          -l /var/APM/logs
        custom terrain path:
                           --terrain-directory /var/APM/terrain
                           -t /var/APM/terrain
root@localhost:/snap/ardupilot/current# bin/arduplane --log-directory /tmp/
Failed to create /var/lib/ardupilot/ArduPlane.stg
root@localhost:/snap/ardupilot/current# bin/arduplane -l /tmp/
Failed to create /var/lib/ardupilot/ArduPlane.stg
root@localhost:/snap/ardupilot/current# touch /tmp/ArduPlane.stg
root@localhost:/snap/ardupilot/current# 
```
`custom log path` does not affect `HAL_BOARD_STORAGE_DIRECTORY`

lucasdemarchi, 2018-03-22:
The .stg files are for parameters, not logs, and is in fact buggy. I'll take a look on fixing this

magicrub, 2018-03-28:
fixed via PR https://github.com/ArduPilot/ardupilot/pull/8027

PR #8027 (2018-05-24): Some Linux updates
Initial intention was just to fix #7983, but then I needed to fix one of my boards to test, derailed on build system, fixed some nitpicks and returned back to what I was supposed to be doing.
closed 2018-05-24 by lucasdemarchi