ArduPilot/ardupilot #7985: Plane: FC sets SAFETY_DISARMED while *in flight*

expntly, 2018-03-21:
#### Issue details
While in AUTO flight, our quadplane suddenly decided to go into SAFETY_DISARMED, which immediately resulted in all outputs to motors being shut off.  This brought the plane down.

[image]

#### Version
3.8.5-beta1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
OctaQuadPlane

#### Hardware type
PH2.1

#### Logs
DF log (102MB):  https://drive.google.com/open?id=1KRORDkRgZlKeNf_cVWoW4VUkl08Mtd33

magicrub, 2018-03-21:
wow, you find all sorts of crazy stuff!

expntly, 2018-03-22:
Definitely flying *a lot*, so that increases the probability of finding strange things.  I'm hoping I'm just misusing the software sometimes.

I took a look at some code under the AP_HAL_PX4 lib, is the only codepath that can set SAFETY_DISARMED literally the switch button being pressed?

magicrub, 2018-03-22:
safety_switch_state() is capable of returning it too

magicrub, 2018-03-22:
and:
get_safety_switch_state()
force_safety_on() from a variety of places such as setup_failsafe_mixing() and GCS_MAVLINK::_set_mode_common()

expntly, 2018-03-22:
Isn't [this in ArduPlane/Log.cpp](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/Log.cpp#L159) literally the only place where STAT.Safety is logged?  From there, I'm taken directly to safety_switch_state(), and it looks like the orb_copy() call returned with safety.safety_off=false.  Then this takes me to PX4 land:

- PX4FMU::pwm_ioctl()
- PX4FMU::safety_check_button()
- PX4FMU::cycle()
- PX4IO::io_handle_status()

and I have no idea what I'm looking at.

magicrub, 2018-03-22:
@tridge ping

peterbarker, 2018-03-22:
On Wed, 21 Mar 2018, Tom Pittenger wrote:

> @tridge ping

Last time this came up it was a physically bad safety switch and vibes 
IIRC.

Safety switches.  You know, the things you lean through the whirling 
blades of death to press?  Yeah.

Naterater, 2018-03-22:
Shouldn't the safety switch be artificially locked in position while plane flying estimate is on?

magicrub, 2018-03-22:
not much of a safety switch if it can't override the motors!

Naterater, 2018-03-22:
not much of a safety feature if it has the potential to bring down a plane unexpectedly. Why isn't the safety switch just an arm/disarm switch then? Not trying to get too theoretical here, but as a newer user, I don't understand the background behind the safety switch in plane. Knowing this moving forward, I'll probably set BRD_SAFETY_MASK to never affect control surfaces. Maybe that should be default. At least with arming, it could be re-armed in the air via GCS if something happened. With a safety switch, you're S.O.L.

magicrub, 2018-03-22:
sometimes people who launch the aircraft are not the ones pressing arm buttons in the GCS so it becomes a safety problem. Having a button is a safety requirement in some scenarios.

Naterater, 2018-03-22:
Agreed, and I don't see why it should be effective while the airplane is flying.

WickedShell, 2018-03-22:
@Naterater because you want it to stop outputs on the vehicle if the autopilot thinks it's flying and it really isn't anymore. BRD_SAFETY_MASK was actually added because I explicitly wanted the autopilot to move surfaces other then throttle during preflight. If you use the overrides with mode manual on anything that has a IO processor (Pixhawk/Pixhawk 2.1) you will always have full control in manual regardless of the safety switch.

Naterater, 2018-03-22:
Thanks @WickedShell. Makes sense, and I think that information should be included in the wiki. Right now the safety switch documentation is functionally correct, but I didn't find anything about its intended setting. Its defaults for plane also don't lead to the same conclusion, and I understand that's difficult because you can set your servoXX_functions to be whatever. I doubt that many users change BRD_SAFETY_MASK to include control surfaces, and it sounds like they should. Sorry to branch off into a documentation issue; I'll go to work.

expntly, 2018-03-22:
FWIW, after witnessing a crash that could have caused a major accident, seriously injured people, and more things than I'm willing to admit, I think it's probably a good idea to make it impossible to go SAFETY_DISARMED while in flight.

Right now, it seems like the logic is

```python
if safety_switch_pressed:
  state = SAFETY_DISARMED
```

But it should probably be

```python
if safety_switched_pressed and is_not_flying:
  state = SAFETY_DISARMED
```

I understand the `is_not_flying` part is tricky.  Though maybe use [isFlyingProbability](https://github.com/ArduPilot/ardupilot/blob/0a08bba4371a58ad56dfc5e1092d9fe19d07069e/ArduPlane/Plane.h#L580) ?

Anyway, for now it'd be great to confirm whether this was actually the root cause?  Would be great to have more devs look into the logs -- I'm happy to provide more info if necessary.

proficnc, 2018-03-22:
I am strictly against the safety button having ANY effect in flight!
It is a single pole, single throw switch that is NOT rated for a task that could allow an aircraft to crash.

In its intended use as a preventative against accidental arming on the ground, it is a two step process... must have safety enabled and be armed... so that’s ok.

But once flying, the safety button should be locked out! 

The current behaviour has caused serious accidents, and I’m greatly concerned about it.

peterbarker, 2018-03-22:
On Thu, 22 Mar 2018, Philip wrote:

> I am strictly against the safety button having ANY effect in flight!
> It is a single pole, single throw switch that is NOT rated for a task that could allow an aircraft to crash.
>
> In its intended use [...]

Should we have an option where you can only disable safety with the 
switch, not enable safety with the switch?

Remember our "is_flying" stuff is only heuristics.  Even the 
weight-on-wheel sensor stuff isn't going to work if you're upside down on 
the ground :-)

WickedShell, 2018-03-22:
@peterbarker step 1 is to actually support the disabled mode. All setting disabled does at the moment is set the starting value of the safety. You can still actually connect  a switch and toggle it regardless of the disable parameter.

Also worth nothing the IO processor has no clue what the safety state is, but it's what is responsible for monitoring the switch status.

magicrub, 2018-03-22:
Using isFlyingProbability() is not correct. Look at is_flying, it checks the probability for you so please stick to is_flying(). 

There's a sticky flag and counter for when you're flying in plane. I'm not sure if this should be full-time or via param but we can do as suggested and add a check for that to lock out the button. However, I'm not sure that would have helped much here. For quadplanes the "is_flying()" flag is mostly disabled because it's easy to trick the logic if you're hovering and we were getting false negatives. So, that would need to get tweaked a little too.

proficnc, 2018-03-22:
“Should we have an option where you can only disable safety with the switch, not enable safety with the switch?”
I like that as an option!

proficnc, 2018-03-22:
+1 Tom! Love it!

peterbarker, 2018-03-23:
On Thu, 22 Mar 2018, Tom Pittenger wrote:

> Well, technically the param BRD_SAFETY_ENABLE is described as "Enable use of safety arming switch" so it implies it is not a disarm switch.

> 0: DISABLED
> 1: ARM/DISARM
> 2: ARM-only

Counterproposal:

This parameter should be renamed to BRD_SAFETY_ONBOOT
0: DISABLED
1: ENABLED

We should add another parameter, BRD_SAFETY_BEHAV
0: ARM/DISARM
1: ARMONLY
2: DISABLED
3: DISPENSECHOCOLATE

peterbarker, 2018-03-23:
On Fri, 23 Mar 2018, Nathan E wrote:

> My proposal:
> Get rid of the is_dangerous safety switch states and "safety switch" completely and change the button to be an ARMING_BUTTON where
> 0: Disabled
> 1: Arming only
> 2: Arm or Disarm
> 3: Disarm only
> 4: DISPENSEMORECHOCOLATE

This doesn't cover the case where you want the vehicle to have safety off 
on boot but disarm only.  Thus my splitting into two params.

Naterater, 2018-03-23:
> This doesn't cover the case where you want the vehicle to have safety off
on boot but disarm only.

So Change BRD_SAFETYENABLE to BRD_ARMENABLE or ARMING_ONBOOT to allow the vehicle to be armed on boot. Then set ARMING_BUTTON: 3 Disarm?

magicrub, 2018-03-23:
BRD_SAFETY_BTN

expntly, 2018-03-23:
It sounds like at the hardware level, this button cannot be trusted. Regardless of whether it is to go from ON to OFF or the other way around.  If this is correct then the arm-only option makes no sense (what will prevent the thing from “arming” behind my back?).  And if this is correct, then the button might not work when my vehicle just crashed and I want to cut power to the motors using that button.  So why have this button at all if it can’t be relied on?  Or am I confused and it does provide *guarantees* in one way?

proficnc, 2018-03-23:
It is not designed to be a trusted input... to do that would require a lot more hardware and software... 

However, as there is visual feedback on the ground, and it’s a two stage workflow...

1. Press safety button 
  a. Light goes solid...
2. Arm vehicle 
  b. Vehicle armed... ( in copter slow spin shows this...

It protects from either stage failing. 

The complications come in when we allow it to be armed, but no safety button pressed... etc.. too many states.

Naterater, 2018-03-23:
If the button is not to be trusted (as stated by the manufacturer AND anecdotal evidence), then I think the term "safety" ought to be removed.

Currently I see two, maybe three use-cases:
1. The first step of a two-step arming process
2. A button to circumvent traditional arming
3. I don't understand wanting the safety off on boot but disarm only. Help me out

To me it's quite simply a pre-arm or an arm button to avoid chopping up things on the ground. The solution would be:
ARMING_BTN
0: Disabled
1: Enabled

I really don't think we should encourage users to disarm using the safety button. Like @peterbarker said:

> Safety switches.  You know, the things you lean through the whirling
blades of death to press?  Yeah.

magicrub, 2018-03-23:
@expntly Are you aware that the button is optional? Set BRD_SAFETYENABLE=0 to disable it.

WickedShell, 2018-03-23:
@magicrub as I indicated above all that does is set the initial state. You can still press the button if it's connected and change the state.

expntly, 2018-03-23:
@WickedShell Thanks for clarifying.  So if I want to eliminate the safety switch, I set that param to 0 and physically remove the button?

The documentation on [BRD_SAFETYENABLE](http://ardupilot.org/plane/docs/parameters.html#brd-safetyenable-enable-use-of-safety-arming-switch) says:

```The safety state can also be controlled in software using a MAVLink message.```

This sounds like it could potentially have happened in this flight?  Does MP have this capability @meee1 ?  I'd like to make sure we're tracking the root cause.

magicrub, 2018-03-23:
@expntly a tlog would show that. Do you have it for this flight?

expntly, 2018-03-23:
tlog: https://drive.google.com/open?id=1gp9T9A-OBPLJb86Ejvam_--HTK3ihrl1

What's the best way to extract the mavlink messages?  I know MP will replay but I'm obviously looking for that mavlink message in particular, not "GUI replay".  (And MAVExplorer.py never worked: "Error loading file" on OSX and Win10.)

magicrub, 2018-03-26:
New issue to track the safety button change https://github.com/ArduPilot/ardupilot/issues/8016

magicrub, 2018-03-26:
I'll close this in favor of the new issue which is tracking the correct problem.

expntly, 2018-04-03:
Thanks!

Re documentation of BRD_SAFETYENABLE that says:

```The safety state can also be controlled in software using a MAVLink message.```

What message am I looking for in the TLOG?  I found I can modify mavextract.py to print the mavlink messages but there's lots of stuff, it would help to know what is that MAVLink message that can control the safety state.

expntly, 2018-04-05:
@magicrub by any chance?  Thanks

magicrub, 2018-04-05:
@expntly oh sorry, I looked this up 3 days ago when you asked but forgot to respond. I looked in the GCS libraries and I don't see any messages that will trigger this. Maybe @peterbarker knows something?

peterbarker, 2018-04-06:
`SET_MODE`

https://github.com/ardupilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_arm.py#L105

LeonDwyer77, 2018-04-12:
Hi All.
I just had Safe mode activate in the air.  That turned 3 grand of gear into a million bits. Including modified cameras and a bunch of hard to replace gear. Woe is me!

Currently, we have 10 aircraft across AUS running Arduplane, some of them in hi-risk environments.
After reading over this thread, I'm going to have to ground the fleet. 

It is inconceivable to me that there is even a discussion about whether or not an aircraft can have all its control shut off while in flight. Or what to call the Safty Switch parameter since its no longer safe. 

This is a software glitch, Not a hardware glitch, given that it's pretty hard to physically activate a switch on an aircraft that is flying at 80m/15mps. I think its safe to say when those parameters are meet the grounding of an IO pin should be ignored. 

A warning needs to be made public as this is incredibly dangerous and the bug fixed as soon as posable.
In this incident, I was lucky I didn't have a 5kg aircraft fly into a car on the highway and cause a fatal accident.  

Tlogs and BIN I'll send to those that want them.

tridge, 2018-04-12:
@LeonDwyer77 yes, I would like both tlog and bin file, thanks.

tridge, 2018-04-12:
I'm working on a change to the IO firmware to offer new options for BRD_SAFETYENABLE this morning, I'd still like the logs though so I can confirm what happened

LeonDwyer77, 2018-04-12:
Thanks for looking at this Tridge.
[2018-04-12 11-19-02.tlog.zip](https://github.com/ArduPilot/ardupilot/files/1905208/2018-04-12.11-19-02.tlog.zip)
[00000016.BIN.zip](https://github.com/ArduPilot/ardupilot/files/1905193/00000016.BIN.zip)

My suggestion would be to mask the function of the button outside of 10meters from launch. Unless Land has been detected or a crash has been detected.   It's not all that elegant, ITs not 100% either But I would imagine it would be effective until a better solution becomes available.

tridge, 2018-04-13:
@LeonDwyer77 your tlog does confirm the safety did engage in flight.
Can you tell me your exact hardware setup? What FC board? If a cube, what carrier board? Were there any changes from the standard setup of the safety switch?

tridge, 2018-04-13:
also, was the switch mounted, or was it loose in the fuselage?

LeonDwyer77, 2018-04-13:
I was a believer 4.8kg.  with a Cube and a here GPS (new batch 'here' with the clear plastic over the LEDs )  I have noticed that these switches are a bit "sticky" compared to the old here GPS with the frosted plastic, and nowhere as good as the stock standard old switches. 

Given the aircraft is very powerful and was in full climb out I'm quite sure that the switch was grounded.

The thing is, it should, in this case, have been ignored.

tridge, 2018-04-13:
I am going to make a software change, I'm just trying to understand the circumstances fully

proficnc, 2018-04-13:
The vague possibility that this can happen frightens me greatly.

The single point switch is NOT SAFETY RATED and as such should NEVER be allowed to disable an aircraft while in flight!!! 

There are so many reasons why this may trigger in flight, but none of these reasons are acceptable as a cause of failure for an Aircraft.

The only safe option is to ignore the button after launch.

LeonDwyer77, 2018-04-13:
And then listen to it after landing or crash.

tridge, 2018-04-13:
@proficnc I am going to give more options for handling the safety switch, but I also think you should look into possible hw faults on the here GPS.

> The only safe option is to ignore the button after launch.

the problem is defining what "after launch" means. Knowing if an aircraft is actually flying is purely heuristic based and is not something you want to use for a safety related feature.

LeonDwyer77, 2018-04-13:
The aircraft does not know its flying?

tridge, 2018-04-13:
like all flight controllers we have heuristics for "is flying". These heuristics take into account many factors like airspeed, ground speed, arming state etc. They can be fooled however.
For example, aircraft could be crashed on the ground in high wind, with motor running and it thinks it is "flying" as it sees high airspeed and its armed. The operator may expect to be able to press the safety switch to make it safe. That is actually the original purpose of the button.
So we try to avoid making safety-related software decisions based on these heuristics.
We are trying to come up with a software change now that avoids this issue, if possible without using the "is flying" heuristic. We are also trying to decide if we should change the default behavior or if we should only offer the new behavior as an option.
We've been discussing this for quite some time. The details are quite involved.

LeonDwyer77, 2018-04-13:
My 2 sense.  Sorry if this sounds frustrated I appreciate your help on this, I just don't think the severity of the situation is sinking in. 

If you know this is an issue, and provide a fix (thanks) but don't change the default behaviour you leave your self-open to being  negligent and there for culpable for an incident.  

The issue here is, this safety feature may cause a crash in the attempt of trying to make an aircraft safe after a crash?  The irony is blindingly bright on this one.

tridge, 2018-04-13:
I do well understand the severity of the issue. The problem is dealing with competing safety concerns. Being always able to use the "safety" switch to disable spinning motors is one side. Not disabling spinning motors when trying to fly is another.

> If you know this is an issue, and provide a fix (thanks) but don't change the default behaviour you leave your self-open to being negligent and there for culpable for an incident.

We do my best to make a sensible set of software changes to try to address this, but in no way do we accept culpability. Do not ever fly ArduPilot in a place where you are risking human life. We do our best to provide great software but we do not ever accept any culpability for any injury that may happen regardless of whether it is a software fault or not. We could not survive as a free software project if we had a legal liability for crashes involving our software.

If you want a warranty that allows you to assign legal culpability then please see if you can find a vendor who will give that to you or seek your own insurance.

Now I would like to go back to working on the software change and stop talking about legal liability. Otherwise I'll have to stop talking to you and instead refer you to lawyers to discuss further.

LeonDwyer77, 2018-04-13:
Yeah point taken on the culpable comment, but how long has the issue been known? anyone digging can see this issue has been known for quite some time and not acted on, and people using the system have not been alerted.  

Anyway, I don't want to argue all this I would much prefer to support you in finding a fix as soon as posable and making sure people know that there is a critical software update that will further protect their aircraft from falling out of the sky.

LeonDwyer77, 2018-04-13:
If you need more resources to make these changes Tridge, AUAV will be happy to compensate you for a reasonable amount of time.

tridge, 2018-04-13:
> Yeah point taken on the culpable comment, but how long has the issue been known? anyone digging can see this issue has been known for quite some time and not acted on, and people using the system have not been alerted.

The problem has only happened once before that I know of (before the report from @expntly), and my understanding was that in that case the user had replaced the standard safety button with a new button of their own design. That in my opinion made it a less critical issue. As far as I am aware this is the first time anyone has seen this happen with "standard" hardware. The behavior of that switch in software has not changed in 5 years.

estebanavv, 2018-04-13:
My 2c. Every setting that affects a behavior globally makes it a lot harder to test all conditions and the interaction among them, furthermore this could risk ground operators. 
If the issue is caused by faulty hardware then a warning should be issued in the documentation.
Statistically an out of control flying craft is very dangerous, but the chances of someone approaching an armed craft on the ground are far greater.

tridge, 2018-04-13:
@LeonDwyer77 did you have a companion computer on board the plane? Or a 2nd telemetry link? The reason I ask is that the SR parameters seem to indicate that more than one link may have been active. 
If you did have a companion computer or a 2nd telemetry link then please describe the setup in detail, and if possible provide logs (tlogs) of the traffic on those links.

tridge, 2018-04-14:
@expntly was your flight in rain? Or any other source of moisture around?

[4 more replies]