ArduPilot/ardupilot #7987: Rover: increase main loop to 100hz or 400hz

rmackay9, 2018-03-22:
Rover current updates at 50hz which is fine for most vehicles and matches the update rates of servos but increasing the rate has two advantages:

- for smaller vehicles and eventually balance bots, a higher update will likely lead to better performance.
- increasing to at least 100hz will mean the EKF runs at the same rate as plane and copter which could make EKF tuning easier (I think, @priseborough has told me something similar to this in the past).

priseborough, 2018-03-22:
Yes, running at 100Hz will mean the same 100Hz covariance prediction rate regardless of platform which eliminates it as a factor affecting tuning.

HomeLH, 2019-12-24:
@rmackay9. I am focusing on balance bots. I try to increase the main loop rate. When the main loop rate is only 50Hz, I cannot understand why the element of the scheduler list can be setted to 400Hz.