ArduPilot/ardupilot #7996: Toshiba LED

shaddow501, 2018-03-23:
#### Issue details
Toshiba LED doesnt work from firmware  AC3.52

at version AC 3.4.6 I have had the Toshiba LED working.
monitoring the I2C outout using arduino I see that the commands printed are: 
0x8381 : 0
0x8182 : 1
0x8283 : 1
0x8381 : 0
0x8182 : 0
0x8283 : 0
0x8381 : 0
0x8182 : 1

On the newer versions from AC3.5.2 I have only printed out those commands:

0:0
0:1
1:1

What did you change and how can I get it to work?

I use Arduino to take the values and convert them to WS2812.

How can I fix it?

shaddow501, 2018-03-26:
Really? No one knows?

rmackay9, 2018-03-26:
Support doesn't really belong on the issues list.  Sorry to be strict but otherwise the to-do list just becomes another support forum.

The toshiba LED certainly works for most people with Copter-3.5.x so it's probably a configuration issue or perhaps the I/O board firmware is out of date.  If you have a safety switch, perhaps try holding it down as the board is powered up.

I wonder if

closed 2018-03-26 by rmackay9

shaddow501, 2018-03-26:
Dear rmackay9

Thanks for your answer,

Can you please tell me what could be related to the Toshiba LED since it did work at version 3.4.6 and doesnt work in version 3.5.5. so It must have been due to some changes.

Can you tell me what or where to look for this info?