ArduPilot/ardupilot #7997: Unable to set `EK3_GPS_TYPE = 0` when using mavlink HIL_GPS message

vincekurtz, 2018-03-23:
#### Issue details
I am using a mavlink HIL_GPS message to send (fake) GPS data to the Ardupilot. Attempting to set EK3_GPS_TYPE to 0 results in the error: EK3: Changed EK3_GPS_TYPE to 1. The HIL_GPS message is generated from an Optitrack motion capture system via the [mavros fake_gps plugin](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/fake_gps.cpp), and does include accurate 3D velocity information. 

#### Version
3.6-dev

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quad-X

#### Hardware type
pixhawk 2.1

vincekurtz, 2018-03-23:
If there isn't a particular reason for this behavior (assuming there is no vertical velocity from a HIL_GPS message), it should be an easy fix. I was able to successfully set EK3_GPS_TYPE to 0 after simply adding

`state.have_vertical_velocity = true;`

to line 135 of `libraries/AP_GPS/AP_GPS_MAV.cpp`.

peterbarker, 2018-03-23:
On Thu, 22 Mar 2018, vincekurtz wrote:

> EK3_GPS_TYPE to 1. The HIL_GPS message is generated from an Optitrack motion capture system via the mavros fake_gps plugin, and does include accurate 3D

Sorry, unrelated to your question here, but we're trying to add support 
into ArduPilot for more appropriate mavlink messages than HIL_GPS.

In particular, https://github.com/ArduPilot/ardupilot/pull/7980 add 
support for four different /position/ message types.

Are you able to test that support for us?

The fact that you're able to feed in velocities leads me to think we might 
need to support VISION_SPEED_ESTIMATE too....

vincekurtz, 2018-03-23:
The original reason for using the HIL_GPS message was that despite its limitations, it's supported by mavros more or less out of the box. I'm definitely open to testing some better solutions though. 

What would be most useful for our use case would be to get position/orientation information from a generic [ROS geometry_msgs/Pose](http://docs.ros.org/api/geometry_msgs/html/msg/Pose.html) topic to the flight controller. I'm not sure if that's exactly what you have in mind with this new support, but I'd be glad help test things out wherever possible. Just let me know what would be helpful.

vincekurtz, 2018-04-02:
Requiring a non-zero value does work for my case, but it's a bit awkward. 

When I first start the stream of HIL_GPS messages (via mavros), EK3_GPS_TYPE gets changed to 1, suggesting zero vertical velocity. If I then go back and change EK3_GPS_TYPE to 0 it stays set to 0.

closed 2018-04-10 by OXINARF

Morris-ae, 2022-10-07:
hi, sorry I couldn't find EK3_GPS_TYPE in the new AC 4.2.3, I am trying to use Marvelmind indoor GPS, is it renamed? thanks in advance

PR #8090 (2018-04-10): AP_GPS: Allow HIL_GPS to flag vertical velocity
Closes #7997
commit 362fb470 (2018-04-10): AP_GPS: Allow HIL_GPS to flag vertical velocity

Closes #7997