ArduPilot/ardupilot #8001: I2C SRF02 driver

rayanPrish, 2018-03-23:
Is the I2C driver automatically used by ardupilot or I need to add them anywhere.
Because I have connected an US (SRF02) with I2C, I haven't write any driver because I think it is like maxbotix i2c driver no ?

muramura, 2018-03-23:
The specifications of this device and SRF 02 are different.
This driver uses 0-1 value.
SRF 02 uses the value of 2-3.
I have PR this driver.

https://github.com/ArduPilot/ardupilot/pull/7423

rayanPrish, 2018-03-23:
thanks I will try it

rayanPrish, 2018-03-23:
I get an error on function MAV_DISTANCE_SENSOR _get_mav_distance_sensor_type() 
[image]

muramura, 2018-03-23:
This PR is based on the revision "a64aee1c".
I have not built it using the master since this date.
The build check at this time, the test is cleared.
It seems necessary to change so that it can be built with the current master.

muramura, 2018-03-24:
I merged the PR source and updated it.
The build finished normally.
I wrote this firmware in PixFALCON and tested it.
The result got the distance normally.

I think. Perhaps, I made a copy. . . When.
Please add a file of the class of SRF 02 and add only the changes to the existing file.

[image]

rayanPrish, 2018-03-24:
sorry I don't understand what you say. I have added the SRF02.cpp & SRF02.h and also 1 case in RangeFinder.cpp

muramura, 2018-03-24:
０。SRF02.cpp & SRF02.h and also 1 case 
       OK

Please do the following.

ardupilot/libraries/AP_RangeFinder/RangeFinder.cpp
１。 Add Include File.
        #include "AP_RangeFinder_SRF02I2C.h"
２。Select Device add (No.17) : 4 places
        @Values: 0:None,1:Analog,2:MaxbotixI2C,3:LidarLiteV2-I2C,5:PX4-PWM,6:BBB-PRU,7:LightWareI2C,8:LightWareSerial,9:Bebop,10:MAVLink,11:uLanding,12:LeddarOne,13:MaxbotixSerial,14:TeraRangerI2C,15:LidarLiteV3-I2C,16:VL53L0X,17:SRF02I2C
 
ardupilot/libraries/AP_RangeFinder/RangeFinder.h
１。enum RangeFinder_Type add
        RangeFinder_TYPE_SRF02I2C = 17,

rayanPrish, 2018-03-24:
this is what i do exactly but I get the same error about MAV_DISTANCE_SENSOR _get_mav_distance_sensor_type().
I compile for OcPoc_zynq:
1_  ./waf configure --board ocpoc_zynq
2_ ./waf --targets bin/arducopter

muramura, 2018-03-24:
I succeed in my environment.

Waf: Leaving directory `/home/murata/work/ardupilot/build/ocpoc_zynq'

BUILD SUMMARY
Build directory: /home/murata/work/ardupilot/build/ocpoc_zynq
Target          Text    Data  BSS    Total  
--------------------------------------------
bin/arducopter  964926  1156  38792  1004874

Build commands will be stored in build/ocpoc_zynq/compile_commands.json
'build' finished successfully (1m34.411s)

rayanPrish, 2018-03-24:
sorry I remember that I have downloaded a ardupilot version for OcPoC maybe this is the problem.
https://github.com/Aerotenna/APM_OcPoC_Cyclone

muramura, 2018-03-24:
Are you running "git submodule update --init --recursive" below?
Because the mavlink relation is under modules.

http://ardupilot.org/dev/docs/git-clone.html

git clone https://github.com/ArduPilot/ardupilot
cd ardupilot
git submodule update --init --recursive

rayanPrish, 2018-03-24:
I have done It. I will retry it

rayanPrish, 2018-03-24:
don't know why but there is no RangeFinder n° 17 (SRF02I2C) 
[image]

muramura, 2018-03-25:
1. FULL PARAMETER LIST BUTTON CLICK 
2. SEARCH "RNG"
3. Display List, set parameters.

rayanPrish, 2018-03-25:
don't work ^^. but I have introduced a hal.console->printf to debug in update() method but it's never displayed in console.

muramura, 2018-03-25:
Have you set RNGFND_TYPE to 17?
Look at this value to determine the device.
Have you set RNGFND_ADDR to 112?
Since the default address of SRF02 is 0x70.

rayanPrish, 2018-03-27:
Thank  you the problem was on my I2C PORT @, I have no message but with a console debug it works perfectly.
Another question If I want to use an other library how can I call it, for example AP_ADC

muramura, 2018-03-27:
http://ardupilot.org/copter/docs/common-rangefinder-maxbotix-analog.html
This information is pixhawk1 connector ADC in analog device..

rayanPrish, 2018-03-29:
okey thanks. why ardupilot try to instantiate always the SRF02 after I have changed it with MAXBOTIX

muramura, 2018-03-29:
http://ardupilot.org/copter/docs/common-rangefinder-maxbotixi2c.html
This url checked...

Perhaps the value of type is 17.