ArduPilot/ardupilot #8004: Trying to build on Windows 7 Environment

shaddow501, 2018-03-23:
Hello all.

I tried to follow your instructions of setting the environment on Windows 7 based PC.

Is there something I have missed??

Checking modules
modules/gbenchmark/.git missing - need module init
+ git submodule init
fatal: Not a git repository (or any of the parent directories): .git
+ echo 'git submodule init failed'
git submodule init failed
+ git submodule status --recursive
fatal: Not a git repository (or any of the parent directories): .git
+ exit 1
make: *** [CHECK_MODULES] Error 1

From your instructions after I:

1. Install the Pixhawk Toolchain by downloading and running the pixhawk_toolchain_installer_latest.exe
2. If you wish to use Eclipse for editing and building, follow these instructions
3. To run MAVProxy locally on Windows follow the MAVProxy’s install instructions for Windows

Even though there is no explanation of what to do with the MAVProxy just to double click on it and that it.

Problems I have encountered with are:
after downloading the Ardupilot ZIP from Github the modules were missing.

I mean the directories are there but they are empty, I didnt see anywhere in your instructions to add them???
So I have added each one, and finally when I thought OK seems good to go, I have encountered with the problem above that I have no clue what to do with it.

Any thoughts?

amilcarlucas, 2018-04-07:
those instructions are really lacking.... sorry for that.

please clone (no zip) the full ardupilot source code from github using:
```
git clone git://github.com/ArduPilot/ardupilot.git
cd ardupilot
git submodule update --init --recursive
```

after that it should work fine.

And if you want to build the SITL: http://ardupilot.org/dev/docs/sitl-native-on-windows.html

WickedShell, 2018-09-21:
Closing as this is a support issue, and has not seen any activity since April. If you need more support please go to https://discuss.ardupilot.org

closed 2018-09-21 by WickedShell