ArduPilot/ardupilot #8009: Flap_Auto and Flaperons broken in Full House wing

Ryanf55, 2018-03-25:
#### Issue details
The airframe has four wing servos, each on their own servo output port on the pixhawk. 
Channel list:
Ch1 : Right Ail
Ch2 : Left Ail
Ch7: Right Flap
Ch8: Left Flap

Goal: 
1) Achieve full span ailerons, with 30% throw on the inner flap surfaces
2) Achieve flaps on the inner flap surface.

Issue: 
On the flap outputs, only one flap works. The other stays at trim value. I have notices this as well setting the output to flaperon, but will only focus on the flap output here.

How to produce the problem:
Enable Aileron outputs on two channels.
Enable "Flap-Auto" to channels 7 and 8
Enable Flap_in_channel to channel #5.
Ensure mode is manual.
Toggle 3 position flap input switch, and observe only 1 flap output.

#### Version
v3.80

#### Platform
 Plane

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Conventional plane. 4 servo wing, dual elevators, single rudder, single motor

#### Hardware type
Pixhawk

#### Logs
[image]

note: I could not attach the .param file. Please advise. For now, see below:
#NOTE: 3/25/2018 12:02:31 AM Plane: Recon
ACRO_LOCKING,0
ACRO_PITCH_RATE,180
ACRO_ROLL_RATE,180
ADSB_ENABLE,0
AFS_ENABLE,0
AHRS_COMP_BETA,0.1
AHRS_EKF_TYPE,2
AHRS_GPS_GAIN,1
AHRS_GPS_MINSATS,6
AHRS_GPS_USE,1
AHRS_ORIENTATION,20
AHRS_RP_P,0.2
AHRS_TRIM_X,-0.06196698
AHRS_TRIM_Y,-0.0217311
AHRS_TRIM_Z,0
AHRS_WIND_MAX,0
AHRS_YAW_P,0.2
ALT_CTRL_ALG,0
ALT_HOLD_FBWCM,0
ALT_HOLD_RTL,10000
ALT_OFFSET,0
ARMING_ACCTHRESH,0.75
ARMING_CHECK,0
ARMING_REQUIRE,1
ARMING_RUDDER,1
ARMING_VOLT_MIN,0
ARMING_VOLT2_MIN,0
ARSPD_AUTOCAL,0
ARSPD_BUS,1
ARSPD_FBW_MAX,29
[... 765 lines cut ...]
TKOFF_THR_MINACC,0
TKOFF_THR_MINSPD,0
TKOFF_THR_SLEW,0
TRIM_ARSPD_CM,1100
TRIM_AUTO,0
TRIM_PITCH_CD,0
TRIM_RC_AT_START,0
TRIM_THROTTLE,45
TUNE_CHAN,0
TUNE_CHAN_MAX,2000
TUNE_CHAN_MIN,1000
TUNE_ERR_THRESH,0.15
TUNE_MODE_REVERT,1
TUNE_PARAM,0
TUNE_RANGE,2
TUNE_SELECTOR,0
USE_REV_THRUST,2
WP_LOITER_RAD,60
WP_MAX_RADIUS,0
WP_RADIUS,90
YAW2SRV_DAMP,0
YAW2SRV_IMAX,1500
YAW2SRV_INT,0
YAW2SRV_RLL,1
YAW2SRV_SLIP,0

Ryanf55, 2018-04-02:
Hello,

It's been a week. I was curious if anyone could help me out, maybe showing where the flaperon mixer code is so I can get a better understanding how it works. 

Thanks!

magicrub, 2018-04-02:
Hi @Ryanf55 sorry we haven't responded here yet. If you want to get a jump start on here, the flaperon code is in a couple places. The flaperon code is not very organized, it's sprinkled into the mixer. Basically, just search the code for "flaperon":
trims: https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/radio.cpp#L331

The big [set_servos()](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/servos.cpp#L524) function does a bunch of last-minute tweaks that are different depending on flight modes. It calls [set_servo_flaps()](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/servos.cpp#L397) which computes the flaps stuff but then calls [flaperon_update()](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/servos.cpp#L169) which adjusts it further to add the flaperon on top of it. They all cascade.

Ryanf55, 2018-04-02:
Thanks!

I noticed that the non-working flap had a very binary approach. It would be pinned to one side depending on whether I had the flaps up or down. Would you like me to take a video of it?

magicrub, 2018-04-02:
@Ryanf55 yes, sounds like a video will help.

Ryanf55, 2018-04-09:
Uploading videos now. I deduced a large part of the problems to this:
When changing servoN_Function's value, a restart of the pixhawk is required for full changes to take effect. 

I was able to successfully get servoN working for the flaps in both flaps (servoN_function = 3) and flaperons (servoN_function = 24/25), however the rates are a bit odd. I'd like to request some additional logic be implemented. If you think I should make this next part a separate feature request, I will.

1)When servoN_function is set to 3, make the servoN_trim be 50% flap value. Dividing the max and min up into 1 range works, but could be better by using the trim value as a 50% marker. This would not require much change, however it would make the flaps travel down in a more synced path. Ideally, the flaps are on a custom curve, but that implementation would be too complicated for most users. 

2) On the flaperon code, add a range that specifies the amount of aileron to mix. Basically, change from using servoN_max and min to another set of values. This is useful on a full house wing when full span ailerons are desired on the inner flaps, however a truncated range of throw is used for ailerons compared to full flaps.

Video 1:
Overview of plane in passthrough and demo of how the flaps should function
https://youtu.be/HUT4G3MPD8k

Video 2:
Flaps in mode 2 with zero output. This is prior to a restart.
https://youtu.be/8xb5su8XfZ4

Video 3:
processing

Video 4:
processing

Naterater, 2019-01-05:
@Ryanf55, Considering the original post title and most of the general issues here have been solved, are you alright if I close this? Crow flap features have also been recently merged, so you may want to check them out.

If you would like to continue the discussion of code issues and limitations, I think these topics might be a good discussion place:  https://github.com/ArduPilot/ardupilot/issues/10036

Ryanf55, 2019-01-12:
Sure you can close it. I think one of the important takeaways is the documentation could be updated to tell people to restart the autopilot after writing SERVON_FUNCTION values.

closed 2019-01-12 by Ryanf55