ArduPilot/ardupilot #8013: Plane/TECS: incorrect handling of _TAS_rate_dem produces significant noise in TECS signal flow.

lekston, 2018-03-25:
#### Issue details
When the airspeed demand input to TECS is varied in time (eg. when flying against strong wind with enabled MIN_GNDSPEED_CM or then using throttle nudge) the airspeed input processing in AP_TECS::_update_speed_demand(void) incorrectly applies the _TAS_dem rate limit. This in certain use case scenarios can result in creating unnecessary oscillations in in both pitch and throttle output signals from TECS. One older example of such oscillations is visible on the below image, another (from SITL built on bleeding-edge master) is provided in the attached Log.

[image]

#### Version
3.8.4 & 3.9

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

#### Logs
[00000044.BIN.zip](https://github.com/ArduPilot/ardupilot/files/1845788/00000044.BIN.zip)

magicrub, 2018-03-26:
pull request is at https://github.com/ArduPilot/ardupilot/pull/8014

IamPete1, 2021-08-30:
Sounds like this is fixed, https://github.com/ArduPilot/ardupilot/pull/8014#issuecomment-485599558

closed 2021-08-30 by IamPete1