ArduPilot/ardupilot #8018: Copter: move get_default_auto_yaw_mode from mode_auto.cpp to mode.cpp.

rmackay9, 2018-03-27:
At the very [bottom of mode_auto.cpp](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode_auto.cpp#L1882) we have a few functions that are part of the Copter class than we should be able to move to the Mode class (i.e mode.h, mode.cpp) or move to Attitude.cpp (@peterbarker do you know where they belong?).

- get_default_auto_yaw_mode
- set_auto_yaw_mode
- set_auto_yaw_rate
- get_auto_heading
- get_auto_yaw_rate_cds

- set_auto_yaw_look_at_heading  <-- called from GCS_MAVLink.cpp
- set_auto_yaw_roi <-- called from GCS_Mavlink.cpp

peterbarker, 2018-03-27:
@rmackay9 Yes, I know how this stuff should be cleaned up (IMO): https://github.com/ArduPilot/ardupilot/pull/7458

amilcarlucas, 2018-04-07:
@rmackay9 just approve #7458.
Then I can do some PRs on top of that. ;)

sarthak268, 2018-04-29:
Is this still open? If yes, please assign it to me.

peterbarker, 2018-04-30:
@sarthak268 Thanks for the interest, but these have been dealt with; these have all been collected into `autoyaw.cpp` IIRC.

closed 2018-04-30 by peterbarker