ArduPilot/ardupilot #8020: Separate RTL switch on TX

padvinder95, 2018-03-27:
#### Issue details
I fly under regulations that ask me to have a separate switch on my R/C that will make the RPAS execute RTL, a safety precaution. This has to be separate from the MODE switch: it's a sort of panic button.

I would like to add a parameter, something like “RTL_CH,” that if set to nonzero will make ardupilot execute RTL if that channel goes above 1500, ignoring the separate MODE switch. When RTL_CH goes below 1500, ardupilot reverts to its previous MODE (manual, FBW, continuing AUTO mission…)

#### Version
Currently on Plane 3.7, but I haven't seen the feature in 3.8. I suppose it would also be useful for the other platforms.

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[x] Rover
[x] Submarine

#### Airframe type
Ornithopter

#### Hardware type
Pixhawk 2.4.8

#### Logs
N/A

geofrancis, 2018-03-27:
with a taranis you just set a switch as a chanel over ride that will force RTL regardless of other switch positions. 

another way to do it is set a switch to override your throttle position so it goes below throttle failsafe therefore engaging RTL.

padvinder95, 2018-03-27:
Ah, so you recommend fixing it in the TX, if possible.
To me, the throttle failsafe option is too roundabout: I may want/have to program a different failsafe action there, not necessarily RTL. However, if one's RC can be programmed to override the mode channel by flipping a switch, that would be a great solution.

Are there many TX that can do this? (Mine probably can, but is it worthwhile adding this feature to ardupilot nonetheless, for others?)

geofrancis, 2018-03-27:
what you want is possible on copter but plane doesnt have ch functions so currently you can only do what you want using the transmitter.

rmackay9, 2018-03-27:
[This PR from Peter Barker](https://github.com/ArduPilot/ardupilot/pull/6819) lays the groundwork for all the vehicles to get the auxiliary function switches that copter has.  I'm hoping to push it to master soon but there would still need to be a follow-up to allow Plane to make use of it.

padvinder95, 2018-03-27:
Yes, this can be closed if https://github.com/ArduPilot/ardupilot/pull/6819 is merged.

geofrancis, 2018-03-27:
what radio do you have?

padvinder95, 2018-03-28:
@geofrancis Jeti DS14: it will definitely be possible to program it in the way you suggested.

Naterater, 2019-01-05:
Supposedly this is solved by https://github.com/ArduPilot/ardupilot/pull/8170, however I don't see the RTL option yet. I'm waiting for clarification.

peterbarker, 2019-04-20:
https://github.com/ArduPilot/ardupilot/pull/11009 <- that PR will permit you to have RTL on a switch.

OXINARF, 2019-08-20:
Done by #11009.

closed 2019-08-20 by OXINARF