ArduPilot/ardupilot #8026: camera shutter-button-down (CAM_DURATION) times vary greatly

ecnalubma, 2018-03-28:
#### Issue details
There seems to be a significant amount of error in the length of time the shutter button is pressed, as measured by analysis of RCOU log entries. There is of course some spread in the time between RCOU log entries, however that spread is very much smaller.

For example, [here](https://polyta.pe/ap/button_press_durations.png) is a histogram of the time between "button down" and "button up" events for a flight where CAM_DURATION was set to 6 (i.e. the desired length of button presses is 600,000 microseconds). The peak of the distribution is in roughly the right spot (actually, it's one bucket, or about 40ms, to the right of where it should be) but the spread is huge.

In particular, we are concerned that some button presses appear to be (very) much shorter than requested. Could this be indicative of hardware struggling to keep up with its workload or is there a bug here? (Or, is it invalid to perform this sort of analysis on the basis of RCOU log entries?)

#### Version
3.8.3

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quadplane

#### Hardware type
Pixhawk 2

WickedShell, 2018-03-28:
Could we get a data flash log that exhibits this/data was pulled from?

ecnalubma, 2018-04-28:
Sorry for the delay - we managed to find out what was going on. It turned out that about 1/3 of our dataflash log entries were being dropped. We have no evidence to suggest that shutter press durations were actually wrong, so I'm closing this issue. Sorry for the false alarm.

closed 2018-04-28 by ecnalubma