ArduPilot/ardupilot #8030: MPU-9250 connect to PIXHAWK 2.46 externa SPI not detected

Jasonsiu, 2018-03-28:
I have a PIXHAWK V2.46 board.
I am able to connect to it the external SPI driver on MPU9250. 
I checked the wiring, and it seems to be ok, but MPU9250 cannot driver on. gx3 gy3 gz3 and ax3 ay3 az3 values are zero.
I only connected SPI, Compass I2C connect wires are empty.
Unfortunately, the MPU is not detected by the Pixhawk.
Firmware version I am using: PX4: 0384802e NuttX: 1bcae90b  APM:Copter V3.5.5 (27229c83)
Thanks a lot, any help will be appreciated.

proficnc, 2018-03-28:
What is a pixhawk 2.46? There is no official board with this designation.

Jasonsiu, 2018-03-28:
Sorry.  PX4FMUv2.4.6

Jasonsiu, 2018-03-28:
Sorry. PX4FMUv2.4.6

magicrub, 2018-03-28:
what ArduPilot firmware did you use?

Jasonsiu, 2018-03-29:
@magicrub Firmware version I am using: APM:Copter V3.5.5
wires connected
board           Vcc     SIP_ext_SCK     SIP_ext_MISO    SIP_ext_MOSI    SIP_ext_NCS       GOIP_ext
MPU-9250      Vcc       CLK                MOSI              MISO                  NCS               Notconnect

I tried this connect also fail.
board           Vcc     SIP_ext_SCK     SIP_ext_MISO    SIP_ext_MOSI    SIP_ext_NCS       GOIP_ext
MPU-9250      Vcc       CLK                MOSI              MISO                   Notconnect        NCS

DavidIngraham, 2018-03-29:
You will need to modify this code to make a Pixhawk 1 look for a 9250 on the SPI bus: 
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_InertialSensor/AP_InertialSensor.cpp#L755

Jasonsiu, 2018-03-29:
@DavidIngraham  If I haven't modify the codes,  MPU-9250(IMU&Compass) use the official firmware (for example Copter V3.5.5 ) I need connect it to internal I2C (I2C1) that it will be detect?

auturgy, 2018-03-29:
I’d suggest a custom ChibiOS build, modifying the fmuv2 hwdef.dat to define the 9250 on external spi bus and chip select pin.

Jasonsiu, 2018-03-29:
@auturgy But CUAV Pixhack V3 has 3rd IMU on board. It base on both fmuv2, Why Pixhack V3 3rd IMU and internal compass are detected?

urlu75, 2018-03-30:
could it linked to the #7431 where one onboard compass is not detected on pixhack v3 ?

DavidIngraham, 2018-04-03:
@Jasonsiu It seems that the driver is doing exactly what it is supposed to be doing. On the Pixhawk 1 build, ArduPilot will only look for the MPU9250 on I2C. This is because using devices on an external SPI bus is dangerous because you risk interfering with the main IMUs. I would encourage you to use an external MPU9250 via I2C.

If you have a compelling reason to use SPI, then you will have to modify the code.

closed 2018-04-03 by DavidIngraham