ArduPilot/ardupilot #8031: Add more than 6 transmitter modes

Naterater, 2018-03-28:
#### Issue details
This is a feature request. With PPM and programmable transmitters with telemetry becoming increasingly common, allow access to all flight modes (not just 6) from a RC signal. 

Goals: 
Allow users to keep/migrate their current settings/flight mode order
Allow users to access all of the modes from a programmable transmitter

Suggestion to solve the problem:
Add FLTMODE_MASK
0=Use FLTMODEn parameters
1 through whatever=disable the FLTMODEn parameters and use the modes within the mask

Add FLTMODE_RANGE
=The PWM width in microseconds for each mode - between 20 and 500

Maybe a sanity check could be added so that the number of modes in FLTMODE_MASK multiplied by FLTMODE_RANGE is less or equal to 1,000.

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2018-04-02:
We've actually had this request before but I'm not a huge fan.  I think more than 6 on a single PWM channel is just adding too much confusion and it's not really necessary because there are other ways to set the flight mode:

- use an auxiliary switch (valid for Copter and Rover)
- use the ground stations set-mode command

geofrancis, 2018-04-02:
why not just make the number of modes optional? that way the default of 6 is still there but advanced users can have more, the lack of modes on plane has been a real problem for a long time. "just use a groundstation" is not really a solution to having more modes on the handset.

there are around 15 selectable flight modes on plane and without a laptop connected I can use a grand total of 6. how is that less confusing?

Naterater, 2018-04-02:
@rmackay9 I'm using plane, so I can't use an auxillary switch. It's also a big hassle to have to have a GCS for flight mode changes. It's mainly focused on plane right now, but it would be applicable to all. Quadplanes have 19 possible flightmodes. I think my solution could also replace the 6 FLTMODEn parameters with 2; however, I do see importance in being able to "order" your flight modes based on pwm, so I maintained backwards compatibility.

I don't think the 2 proposed parameters will have any added confusion, and it would be set to 0 (disabled) by default. It's an "advanced user" parameter that is actually pretty simple. No more complicated than BRD_SAFETY_MASK.

Keeping the modes on 1 channel saves channels, and most people are already programming transmitters to get all 6 modes. Why not let them program to get all of the modes?

lvale, 2018-04-02:
So, @Naterater would be better perhaps to limit the scope of this only to plane, because on copter you can use all the option channels for some specific flight mode settings, such as land and RTL

Naterater, 2018-04-02:
@lvale, there are still 21 modes available to copter, and I don't think the list has ever shrunk. Even with a 16 ch transmitter, you can't access all of the modes on the transmitter. I think it would be simpler to keep them all on 1 channel anyway -- you wouldn't have to even think about what happens when you have RTL mode active on one channel option and then you activate the channel option for acro; what happens? Keeping it on a single channel avoids the case altogether.

lvale, 2018-04-02:
@Naterater please don't enumerate all the possible flight modes. I know them. https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/defines.h#L91-L113

You won't use them.

Realistic use of flight modes either on a recreation or a professional mode won't need them and having 21 flight modes on a single pwm channel would be a task that i'm not so sure you could do without some programming on a Taranis, and other radios would be near impossible, not to mention the fact that even if possible selecting a specific flight mode would be very hard....

geofrancis, 2018-04-03:
It's really not that complicated to have as many flight modes as you want on a taranis,  just using my 6 position switch along with a 2 position gives me 12 modes.

geofrancis, 2018-04-03:
The only way I could see this being a problem is if your using a low resolution system with ppm that dithers a little, but only at high numbers of modes 12+

geofrancis, 2018-04-03:
why not do it the way cleanflight derivitives do by just having all modes broken out and the user sets a range and channel for everything. that way you can have as many modes as you want spread out over how many channels you want.

Naterater, 2018-04-03:
To be clear, I am not requesting that the current 6-flight mode system be removed. By default, nothing changes, and you would set your 6 flight modes the traditional way.

By the addition of 2 parameters, FLTMODE_MASK and FLTMODE_RANGE, any/all channels can be activated by a programmable transmitter with as few as 5 channels. This is for advanced users and tinkerers (a significant portion of the community). I think it's the simplest solution. For commercial users, this can be a benefit because you can increase OR decrease the number of flight modes available.

Other points brought up: 
-Allow 2 "banks" of flight modes with a 2-position switch to select the bank. I think this adds too many parameters, and still limits the selectable modes to 12 or 18. I'm not sure I see how it can be done as simply as a mask.
-Instead of a FLTMODE_RANGE, use something like a FLTMODE_NUMBER to set the maximum number of flight modes and automatically divide the PWM range based on that number. I think that it removes some of the simplicity of the range idea. For example, setting the value to 12 would divide the pwm banks into 83ms width, and channel 4 would be from 333 to 417 -- not exactly an intuitive number.
-Have all modes broken out and selectable ranges/channels. I think that it's a parameter nightmare and far more complex than is needed. Maybe this would be accomplished with a nice GUI, but there are a number of GCS that would need to do more design to make it easy.

lvale, 2018-04-03:
Like it was told at the dev call, we already have 6 fixed function channels. Adding more reserved channels would be very detrimental when you require/need/use the Option channels up to 16, and it is very easy even on a simple copter setup to use all 16 channels, and usually you run out of available switches to all the functions.

The multiple bank concept is something that is common with a 2 POS and a 3POS switch to use all 6 flight modes.

I would try to have the options extended to Plane like on Copter, and i'm sure you'll be satisfied with the current Copter/Rover system.

geofrancis, 2018-04-03:
having a dual bank system is going to be more compatible with more radios from a tx programming point of view but its going to use more channels that are all ready in short supply with complicated builds.

avsaase, 2020-10-30:
Did this feature request ever go anywhere? I am just switching from INAV over to ArduPlane and I am finding that having only six modes is a serious limitation. The methods of mode selection in cleanflight/betaflight/INAV is very flexible but still allows for all modes to be selected via a single channel if you use some logical switches in your transmitter.

peterbarker, 2020-11-01:
On Fri, 30 Oct 2020, Alexander van Saase wrote:

> Did this feature request ever go anywhere? I am just switching from INAV over to ArduPlane and I am finding that having only six modes is a serious

It did not.

> limitation. The methods of mode selection in cleanflight/betaflight/INAV is very flexible but still allows for all modes to be selected via a single
> channel if you use some logical switches in your transmitter.

On the face of it this shouldn't be hard to implement.  New parameter 
specifying number of divisions on the mode channel, split into equal 
divisions.

rmackay9, 2020-11-01:
In Copter and Rover at least (I'm less sure about Plane) you can also setup flight modes on an individual switch.

avsaase, 2020-11-02:
> In Copter and Rover at least (I'm less sure about Plane) you can also setup flight modes on an auxiliary switch.

If I understand correctly this is not possible in Plane. Regardless, I prefer to have a mutually exclusive flight modes on a single channel in order not to waste valuable channels.

skorokithakis, 2021-03-07:
I agree that more than 6 modes are necessary, even with the extra mode channels. My use case is that I want 7 modes (which I use on every flight, multiple times), with priorities. This is basically impossible to do unless the mode channel supports more than 6 modes.

I have set up switch overrides on my radio (so each switch produces a specific output on the mode channel, overriding the previous ones), so I can have one mode override another, but adding a second channel introduces race conditions that break this system.

What's more, there's no technical reason why we can't have ten or twenty modes, channel range is free. There's no downside to at least adding more modes as a stopgap until a better mode system comes along.

rmackay9, 2021-03-16:
We discussed this on the dev call and the advice on implementation included:

- we should consolidate the implementation across vehicles into an AP_Mode or AP_FlightMode library
- a _SCHEME parameter could be added to allow choosing the PWM spacing and number of channels
- add a second bank of FLTMODEx parameters

mwerezak, 2023-06-28:
Using auxiliary switches for extra modes seems to work very poorly when you need to switch between modes on different aux switches. Instead of switching, it appears to frequently get stuck on the mode indicated by the main mode switch instead of actually going to the desired mode.

skorokithakis, 2023-06-28:
@mwerezak if it helps, I switched to ArduCustom, where I implemented 12-mode functionality, which works flawlessly:

https://github.com/ArduCustom/ardupilot

rmackay9, 2023-06-28:
@mwerezak,

This sounds like a support issue so it's probably best to report in the forums if possible.  In particular if you're using 4.4-beta then I check the reports quite often.  https://discuss.ardupilot.org/c/arducopter/copter-44/169

timtuxworth, 2023-06-29:
Perhaps but it also re-ups this concern which IMO is valid. There should be a simple consistent way to have more than 6 modes, and having 6-pos switch + random other switches is awkward and confusing.
Ideally either
* ability to specify the number of modes on a channel and the width of each
* a second 2 or 3 position switch that acts as a "bank selector" - giving 12 or 18 modes.
* something similar to the Aerobatic "tricks on a switch" mechanism