ArduPilot/ardupilot #8032: External Toshiba LED

shaddow501, 2018-03-28:
hello 

I have a question.

If the external Toshiba LED worked on Pixracer at firmware version 3.4.6 and stopped working at firmware 3.52~3.55 is it an issue or feature request ?

The External Toshiba LED gets I2C commands as like 0x81/82/83 and so on.
With the new firmware 3.5.2 and above the External Toshiba LED stopped getting those values, I was able to monitor the I2C bus at address 0x55 and now it gets  0 or  1. (not HEX value)

I have no clue where to look for the problem, I hope to get the support to fix this problem. to know where those values are coming from. something does not work right.
But again, I really dont know if it an issue or feature request.
For my understanding is that if something worked before and stopped working with the newer firmware it is an issue, but I might be wrong...

WickedShell, 2018-03-28:
As you were told in #7996 support issues belong on discuss.ardupilot.org rather then in the issues list. Please open an issue there, or post on the Copter 3.5 release thread asking about it.

closed 2018-03-28 by WickedShell

auturgy, 2018-03-29:
What changed was removing the driver from the PX4 HAL and using the driver in the libraries.  This was a broad cleanup removing as any duplicated drivers as possible, so that maintenance of the STM32 and Linux boards could be more common.  You need to go back to the source from 3.4.6, find the driver from the PX4 HAL, then compare it with the current driver, to figure out why your device doesn't work.

shaddow501, 2018-04-03:
This is what I modify and works for me, for anyone that is interested... I run the code on Atiny85 module but you can use Arduino as well just change the TinyWireS lib to Wire lib and modify the "Wire" locations. 

#include <TinyWireS.h>
#include "FastLED.h"

#define NUM_LEDS 4

// Data pin that led data will be written out over
#define DATA_PIN 4
// Clock pin only needed for SPI based chipsets when not using hardware SPI
//#define CLOCK_PIN 5

long lasttimeLED = 0;
#define delayLED 15 //ms
long lasttimeCYC = 0;
#define delayCYC 0 //ms

CRGB leds[NUM_LEDS];
bool thergbled[3] = {0, 0, 0};
bool changeflag = false;
bool i2cflag = false;
#define downrate 1
const int ledPin =  1;
int ledState = LOW; 
int done = 1;

void setup() {
 TinyWireS.begin(0x55);                // join i2c bus with address
 TinyWireS.onReceive(receiveEvent); // register event
//  Serial.begin(9600);           // start serial for output
  FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);  // init LED
//  
}

void loop() {

  LED();
  CYC();

}

void LED() {
  if ((millis() - lasttimeLED) > delayLED) {
    lasttimeLED = millis();

     if(i2cflag) {
      leds[0] = CRGB(255 * thergbled[2], 255 * thergbled[1], 255 * thergbled[0]);
      FastLED.show();
      i2cflag = false;
      changeflag = false; // skip CYC change since already update in this loop
      done = 1;
    }
    else if(changeflag) {
      FastLED.show();
      changeflag = false;
    }
  }
}

void CYC() {
  if ((millis() - lasttimeCYC) > delayCYC) {
    lasttimeCYC = millis();
    for(int i=(NUM_LEDS-1);i>=1;i--) {
      leds[i] = CRGB(leds[i-1].r *downrate , leds[i-1].g *downrate, leds[i-1].b *downrate );
    }
    changeflag = true;
    done = 1;
  }
}

// function that executes whenever data is received from master
// this function is registered as an event, see setup()
void receiveEvent(uint8_t howMany) {
 if( done == 1 ){
  if ( TinyWireS.available() >=2 ) { // loop through all but the last
    int ledreg = TinyWireS.receive(); // receive byte as a regid
    int ledval = TinyWireS.receive();
     int V = (ledreg);
     if( V == 0 ){
      int R = 0;
      int  G = 0;
      int B = 0;
     Serial.print(V);
     Serial.print(" ");
     Serial.print(R);
     Serial.print(" ");
     Serial.print(G);
     Serial.print(" ");
     Serial.println(B);
     Serial.println("end");
 
      thergbled[0] = R;
      thergbled[1] = G;
      thergbled[2] = B;
      i2cflag = true;
      done = 0;
     }
     else
     {
     int R = (ledreg & 0x0F);
     int  G = (ledval);
     int B = (ledval& 0x0F);
     
      Serial.print(V);
     Serial.print(" ");
     Serial.print(R);
     Serial.print(" ");
     Serial.print(G);
     Serial.print(" ");
     Serial.println(B);
     Serial.println("end");
 
      thergbled[0] = R;
      thergbled[1] = G;
      thergbled[2] = B;
      i2cflag = true;
      done = 0;
     }
  }
}
}