ArduPilot/ardupilot #8060: ArduPlane: GCS unable to resume mission when RTL_AUTOLAND != 0

DonLakeFlyer, 2018-04-01:
In order for QGC to support Resume Mission after a battery change it remembers the last waypoint which was current when the vehicle leaves Auto flight mode. I believe this is the same as Mission Planner, but not fully sure. If the vehicle has RTL_AUTOLAND != 0 when the user changes to RTL flight mode, it starts the RTL but it does not exit out of Auto. This in turn leads to the inability to use Resume Mission functionality.

Is there some other way to do this. Seems like an important use case for fixed wing.

QGC Issue here: https://github.com/mavlink/qgroundcontrol/issues/6286

DonLakeFlyer, 2018-04-02:
@WickedShell How do you deal with this?

WickedShell, 2018-04-02:
@DonLakeFlyer I've never set RTL_AUTOLAND to non 0 values so I've never encountered this. :)

The core problem is this mode isn't actually RTL. It is legitimately flying the planned mission after the selected DO_LAND_START, so the reporting of auto is correct (if slightly weird). Reporting RTL would also be confusing as the vehicle isn't executing it's normal RTL behavior. I'm not sure what the best fix for this is.

The best fix I can come up with at the moment is if we could report internal flight stage states. (see https://github.com/mavlink/mavlink/pull/727 for the example idea on `flight_stage` reporting)

magicrub, 2018-04-02:
@DonLakeFlyer The problem is "RTL" is fairly dumb behavior; flies home and loiters. Allowing the user to ave a customized landing sequence is preferred. However, I understand your problem and agree that losing your place is a problem that should be address. Good thing to discuss at the dev call today.

DonLakeFlyer, 2018-04-02:
What time is dev call, not sure if I'll be around.

DonLakeFlyer, 2018-04-02:
Nope I missed it.

DonLakeFlyer, 2018-04-02:
> Reporting RTL would also be confusing as the vehicle isn't executing it's normal RTL behavior.

I would say that from an end user perspective that it is executing RTL. Especially since you specifically told it to RTL. I get that internally to the source that isn't really what is happening. If you RTL a multi-rotor it goes all the way to the ground. If RTL_AUTOLAND != 0 then fixed does the same thing. That seems like RTL to me.

magicrub, 2018-04-02:
Weekly dev call is 2300 UTC (that's 3 hours from now)
http://ardupilot.org/dev/docs/ardupilot-mumble-server.html#weekly-dev-calls

tridge, 2018-04-03:
we will look into reporting mode RTL during RTL_AUTOLAND missions, as long as they don't do a DO_JUMP

DonLakeFlyer, 2018-04-03:
Thanks everyone

Naterater, 2019-01-05:
Was this worked out? I recall the difficulty from jumping to mission items at the end of the mission. There has been discussion about abstracting the autoland sequence if I remember correctly, and there was this thread in the discuss forums. https://discuss.ardupilot.org/t/multiple-closed-loop-mission-flight-plans/28506.