ArduPilot/ardupilot #8064: Plane: Climbing Turns result in power-on stalls

Naterater, 2018-04-02:
#### Issue details
Extracting max performance out of planes means turning up the bank angles and pitch angles. Unfortunately this puts the plane in a dangerous situation when both happen simultaneously (like after takeoff with a waypoint that is downwind). My TECS tuning is correct, and the maximum bank angle is perfectly achievable for level flight.

Maybe something could be added to improve the stall prevention code to include pitch? My STALL_PREVENTION parameter was on, but it did not act quickly enough to prevent the accelerated stall. Most planes just don't have enough power to climb at PITCH_MAX and turn at LIM_ROLL_CD simultaneously, even if they are set well individually. 

KFF_THR2PTCH and TECS_RLL2THR don't do any good if the plane is already climbing at nearly 100% throttle.

#### Version
Plane 3.8.4 & Latest 3.9 (3-31-2018)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
All - 

#### Hardware type
Pixhawk 2.1

#### Logs
N/A - I almost always add manual down pitch during the turns after takeoff.

WickedShell, 2018-06-25:
@Naterater Pitch isn't incorporated into `STALL_PREVENTION` at the moment, only demanded roll is. FWIW I was very very surprised when I found out about that recently.

Naterater, 2018-12-17:
This also has RTH implications at low altitude.