ArduPilot/ardupilot #8067: Baro: store dev_id to catch changes barometer

rmackay9, 2018-04-03:
The AC_Compass library catches changes in a vehicle's compass setup (i.e. compasses added or removed) by storing each compass's device id in a parameter at the time that the compass calibration is done.  Then when the user goes to arm, we check that the device ids stored in the eeprom match the device ids detected after startup.

We could do something similar for barometers which might be useful now that external barometers (using CAN?) are becoming available.  We would need to decide when to store the device ids to eeprom because, unlike compasses which are only calibrated once by the user, barometers are calibrated at startup.

WickedShell, 2019-04-05:
I suspect we could accomplish the primary desire of this by introducing the same concept of expected ID as we have for compasses. It requires the user to have set them of course, but for most users adding an explicit step of "save my baro ID's" just makes the setup more complicated. (And we can't use the existing MAVLink message that starts the calibration because it isn't unreasonable to be sending this every flight, which would cause you to overwrite the old ones)

rmackay9, 2020-01-24:
This would have helped a user in this discussion re Copter-4.0: https://discuss.ardupilot.org/t/external-baro-sensor-on-can/51390/3

The problem is just that users have difficulty figuring out which barometers are being used.

rmackay9, 2020-01-24:
Here's another discussion (separate from the one above) where this would have helped a user: https://discuss.ardupilot.org/t/barometer-glitch-inducing-ekf-primary-change/51345/5

rmackay9, 2020-07-22:
PR https://github.com/ArduPilot/ardupilot/pull/14847 has resolved this

closed 2020-07-22 by rmackay9