ArduPilot/ardupilot #8069: All mavlink channels that have streamrates are streaming

peterbarker, 2018-04-03:
#### Issue details
The default streamrate for ADSB is 5; it is zero for every other stream: https://github.com/ardupilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L555

Having this non-zero means that the channel is marked as streaming (https://github.com/ardupilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Param.cpp#L294) and we'll do a lot more work figuring out what we should send.

I suggest we change the default stream rate for this parameter to 0.

The other option is to eliminate the streaming mask, because it's useless as-is.

#### Version
master

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
All

#### Hardware type
All of them.

#### Logs
n/a

WickedShell, 2018-04-03:
My initial reaction is that we should change the default ADSB stream rate to 0 to reduce workload on unused ports. My follow on reaction was "if I want that, why not set a port to no protocol and just not worry about it".

amilcarlucas, 2018-04-07:
I do set all non-interesting streams to zero, AND disable unused ports (-1).
I think that I should extend the stream rate calculation tool to other vehicles and add more features so that the users start using it more and start being more aware or data-rate limitations.

What features should I add?