ArduPilot/ardupilot #8076: QuadPlane: QRTL moves away from home if vehicle took off before getting GPS lock

rmackay9, 2018-04-04:
I've tested QuadPlane with master (4th Apr 2018) and noticed this odd behaviour if the user takes off before the EKF sets the origin.

- when QRTL is initiated the vehicle originally flies to the takeoff location but then as it descends it moves to another location (perhaps the location that the vehicle was at when the EKF origin was set
- the vehicle will stop descending a meter or two off the ground

WickedShell, 2018-04-04:
This was done with `ARMING_CHECKS 0` and given that plane requires EKF to be happy/using GPS to arm I'm not sure this is a real plane issue to chase until we actually support flight without GPS.

rmackay9, 2018-04-04:
Yes, arming checks were off.  I think it's a real problem though because it just seems unreasonable that the vehicle would initially fly to home and then, as part of the descent fly to another location.  I think it's an issue with the target to the multicopter loiter controller being incorrect somehow.