ArduPilot/ardupilot #8089: Alt Healthy even if out of range high ?

Kiwa21, 2018-04-04:
https://github.com/ArduPilot/ardupilot/blob/3d84dfcc2e8873a49b80b5705c961273a0f03d55/ArduCopter/sensors.cpp#L39

Rangefinder_state.alt_healthy is reported as true even if the current alt is above the rangefinder max distance ?

Should adding  something like this make sense ?

`rangefinder_state.alt_healthy = ((rangefinder.status_orient(ROTATION_PITCH_270) == RangeFinder::RangeFinder_Good) && (rangefinder.range_valid_count_orient(ROTATION_PITCH_270) >= RANGEFINDER_HEALTH_MAX));

int16_t temp_alt = rangefinder.distance_cm_orient(ROTATION_PITCH_270);

if( temp_alt >= rangefinder.max_distance_cm_orient(ROTATION_PITCH_270)){
    rangefinder_state.alt_healthy = false; //out of range high
}
`

Kiwa21, 2018-04-12:
I agree it's might not be the good term.
However, rangefinder alt is used by EKF or other library when alt_healthy is true even if alt > alt_max.

For example : 

```
void Copter::update_precland()
{
    int32_t height_above_ground_cm = current_loc.alt; //using ekf get hagl

    // use range finder altitude if it is valid, else try to get terrain alt  // TODO change that shit
    if (rangefinder_alt_ok()){
        height_above_ground_cm = rangefinder_state.alt_cm;
    } else if (terrain_use()) {
        if (!current_loc.get_alt_cm(Location_Class::ALT_FRAME_ABOVE_TERRAIN, height_above_ground_cm)) {
            height_above_ground_cm = current_loc.alt;
        }
    }
    copter.precLand_sm.set_height_above_ground_cm(height_above_ground_cm);
    copter.precland.update(height_above_ground_cm, rangefinder_alt_ok());
}

```

should be changed to 

```
void Copter::update_precland()
{
    int32_t height_above_ground_cm = current_loc.alt; //using ekf get hagl

    // use range finder altitude if it is valid, else try to get terrain alt  // TODO change that shit
    if (rangefinder_alt_ok()){
        height_above_ground_cm = rangefinder_state.alt_cm;
        **if(height_above_ground_cm >= copter.rangefinder.max_distance_cm_orient(ROTATION_PITCH_270)) { //out of range high
            height_above_ground_cm = current_loc.alt; 
        }**
    } else if (terrain_use()) {
        if (!current_loc.get_alt_cm(Location_Class::ALT_FRAME_ABOVE_TERRAIN, height_above_ground_cm)) {
            height_above_ground_cm = current_loc.alt;
        }
    }
    copter.precLand_sm.set_height_above_ground_cm(height_above_ground_cm);
    copter.precland.update(height_above_ground_cm, rangefinder_alt_ok());
}

```
Rather than a boolean, it may be better to define it as 0 : not healthy / 1  : healty / 2 : out of min range / 3 : out of max range.

growingwings, 2019-06-27:
Not directly related to above but a bit more specific issue. AP_RangeFinder_LeddarOne.cpp is not filtering invalid detection. Average calculation (using both potential invalid data and different reflections) is a bit questionable too.

Leddar One User Guide: "Distance and amplitude are 0 for a detection that is not present (for example if register 23 is 1, registers 26 to 29 will all be 0)."

https://leddartech.com/resources/#leddarone-en/

Could make separate issue too...

IamPete1, 2021-09-04:
Rangefinder `status_orient` will report `OutOfRangeHigh` not `Good` in that case so it will not be marked as healthy. 

Maybe I have misunderstood, unfortunately the code line number in the link is not pointing to anything anymore, please re-open I'm wrong.

closed 2021-09-04 by IamPete1