ArduPilot/ardupilot #8096: Loiter to Alt : tangent cross-track exit

kikislater, 2018-04-06:
#### Issue details
As advise here : https://github.com/mavlink/qgroundcontrol/issues/6137
QGroundcontrol provide a handy tool to autoland a flying wing. Is there a way to have support for tangent cross-track exit in loiter to alt mission mode ? Or any idea how to solve and make this usefull thing working on Arduplane ?

#### Version
3.8.4

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Flying wing

#### Hardware type
Pixhawk

#### Logs
No logs yet but could provide.

magicrub, 2018-04-06:
@kikislater Already implemented via PR https://github.com/ArduPilot/ardupilot/pull/3750 and https://github.com/mavlink/mavlink/pull/531. We just need @DonLakeFlyer to set the flag on the mavlink packet for QGC to take advantage of it

closed 2018-04-06 by magicrub

DonLakeFlyer, 2018-04-06:
Uh, it does. The problem is that the vehicle doesn't always circle the loiter. So it leads to an unpredictable exit point as shown in the graphic.

DonLakeFlyer, 2018-04-06:
All explained in the QGC issue

magicrub, 2018-04-06:
@cody303 @kikislater  @DonLakeFlyer This all definitely works in ArduPlane but there is a corner case that just got fixed here ArduPilot/ardupilot#7934 where if you're using a LOITER_TO_ALT but performing a turn and you're already at the altitude in the first couple seconds of the loiter then the circle is not much of a circle. This was not planned to be in the current 3.8.5beta but I think it would be a good candidate to include.