ArduPilot/ardupilot #8107: Tailsitter: Improve scaling of control surfaces

palm369, 2018-04-07:
#### Issue details
Tridge coded a scaling feature for the control surface outputs of tailsitter aircraft, so the flaps deflact more when you lower throttle. This makes sense, so the output force stays the same.
But there is still a lot to do.

The scaling feature for should also take into account:
- the real airflow over the wing due to (1) Throttle of the (2) Individual motor and (3) airspeed. Right now it only taking (1) into account
- the hover trottle will change hugely from voltage drop of the main battery and if payload is put to the craft. Eg: If you fly with an Lipo Cell then you will start flying at approx 4.0 V and you will drain the cell down to 3.3 V (both under load). This means that you will have only 68% = (3.3² / 4²) of the thrust left.
When using a 18650 cell this will get even worse. So you might end up with only 40% = (2.5²/4²)
- doubling output does not mean that the force of the control surface is doubled. Its less, except if you have very little angles deflaction. (sine function!)

Also the maximum scaling factors should be available as parameters.

Find the code on the very bottom: https://github.com/ArduPilot/ardupilot/blob/plane3.8/ArduPlane/tailsitter.cpp

#### Version
AP 3.8.5
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Tailsitter
#### Hardware type
any

IamPete1, 2018-04-07:
I also am having issues due, I think, to the maximum throttle scaling value, this is currently hard coded to 5. We see D oscillation in the servo outputs at very low throttle. This disappears when a very small amount of throttle is added. see: 

https://www.youtube.com/watch?v=Rmpvpy-yCvI&feature=youtu.be

The aircraft in question has a very low hover thrust currently, due to having no payload and only half the air frame, this causes the throttle scaling to very rapidly change at low throttle, it very quicky goes from 5 to <3. See;  
[image]

In this case it would be useful to be able to change the maximum scaling value as a parameter, I can change the hard-coded value to confirm it is the cause of the oscillation.

as for the other issues:

- I think there will be little difference between left and right motor throttle in flight, and i guess there shouldn't be much airspeed in vertical flight mode. 

- This is all-ready implemented, auto hover thrust leaning can be enabled by setting Q_M_HOVER_LEARN, or thrust can be scaled with voltage by setting Q_M_BAT_VOLT_MIN and Q_M_BAT_VOLT_MAX. although i guess it may not be a good idea to use voltage scaling and hover throttle learning at once

- not really sure how to overcome this, you would need to give the code information about linkages and servo angles

IamPete1, 2018-08-12:
this PR exposed the max scaling factors, https://github.com/ArduPilot/ardupilot/pull/8173, This interpolation PR https://github.com/ArduPilot/ardupilot/pull/9167 does I guess throttle and airspeed, i think individual motor throttle would be very tricky to implement.

I have had a quick play with giving the code info about tilt angles, didn't prove to help very much, probably some advantage to be had tho

IamPete1, 2021-07-26:
There are lots of scaling options in master now, we have min and max limits and the disk loading function takes into account the forward airspeed and throttle level. 

I'm going to close this, but feel free to re-open if you have any ideas for things that we have missed.

closed 2021-07-26 by IamPete1