ArduPilot/ardupilot #8108: QuadPlane: make de-transition smoother in AUTO

expntly, 2018-04-07:
#### Issue details
I don't think I've ever managed to get a smooth de-transition.  I've played with lots of parameters over time.  What I define as a smooth de-transition would be where the plane doesn't pitch up/down, roll, yaw more than 20 degrees in any of these axes.  Also the altitude would remain constant +/- some small delta.

It's not uncommon to see de-transitions lead to over 3Gs in acceleration (in Z mostly) as the vehicle pitches up and climbs.  This leads to unnecessary stress in the airframe and higher risk of control loss.  Also this just makes for really scary de-transitions close to the ground, and almost always having to de-transition super high so that the FC has time to re-stabilize before descending and landing.

I've had both super tiny quadplanes and super large ones and I've always had this "issue".  Adjusting the distance between WP and VTOL_LAND helps a bit but doesn't solve the problem.  Having the two waypoints very close to one another amplifies the issue obviously.

Ideally I can imagine the aircraft flying cruise, turning off the fixed wing throttle and **slowly increasing VTOL propulsion** to keep lift constant.  The vehicle would stay straight and level, and would not climb.

I'm familiar with the new `Q_TRANS_DECEL` param but it is not being used in AUTO as far as I know.

I'm seeing the same thing happen with SITL.

#### Version
Literally all versions since the quadplane support was added.  Lately 3.8.5 beta 1.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
X4 and X8

#### Hardware type
PH1 and PH2.1

expntly, 2018-04-19:
friendly ping @magicrub @tridge @rmackay9

expntly, 2018-05-20:
re friendly ping @magicrub @tridge @rmackay9 

Also, is there anything I can do to help?  Provide more data on what the problem is?  Every de-transition hurts here.  Thanks

magicrub, 2018-05-21:
I'm so tell sorry that we have neglected these quadplane issues. I, myself, have not worked on quadplane. Most of the work has been @tridge and recently @wickedshell has been working on it.. so I would rely on them.

pompecukor, 2018-05-31:
I have the exact same experience as the Arthur. De-transition always end up making my  hear skip 1 or 2 bits. They are hard and aggressive. Almost like if the plane was showing off that it can stop and do a flip at the same time. It always settle fine after, so I am getting to trust it. Also shows that winds will not be a problem.. Still it should not look like this.
Here are a couple of such video. As you can see in all, the fixed wing to VTOL transition, it does like there is not tomorrow, or as if it was impatient. Even the yaw after stoping is very hard (why should it be), then is start to come down fine. PID is spot on. So that is not the issue, also transition to plane is very smooth.

Mostly for me I get very hard pitch up. As if it wants to deep stall stop.

https://youtu.be/QfrL0Dvbb5o

In second part of this second video, you can actually see the same aggressive attitude, when it fact the mission was just VTOL Take off to 10m and then VTOL land  5 meter to the front and side of where it took of from. You can see is sprints forward very hard, then realizes its there and breaks as it if was a pod race in SW1. Then doubles back and to the side to get to the right spot and lands smoothly.
I haven't tested this much, but I think that generally, any auto positions change in VTOL mode acts like this. So Qloiter, Qhover, Qstabilized are all fine and dandy, but if in auto mission and brief position change is needed (except for transition to fixed wing), then I see the same thing.
Dont mind the brass swooping by, they did not come for me. What we do is still legal where i live 💃 
 
https://youtu.be/Fhjrh1Xpfw0

pompecukor, 2018-05-31:
Is that what you are seeing too @expntly  ?

pompecukor, 2018-05-31:
I was wondering and just checked my older POV videos. When I was in FBWA and switch to Qloiter then the transition was smooth. So it first tried to slow down then slowly upped the VTOL motors.

So think the issue here is going to be related to QRTL and VTOL land specifically. So it is not anticipating and gradually stopping as it would do in semi auto more (FBWA to QLoiter), but is flying with full speed and then wants to stop suddenly. No transition time. I think that is why it is pitching up. Even tough I set the radius to 75m, which should be more than enough, it tries to trigger on the spot or something like that. @expntly can you confirm this for yourself too?

I am guessing the more information, the easier it would be to fix....

expntly, 2018-06-24:
We see this in any mode where we go from fixed cruise to VTOL.  We fly mostly AUTO though.  Which is the best way to consistently reproduce anyway.  But taking over from a fixed wing AUTO to QLOITER is a maneuver that we really don't like to do.  In any case, the vehicle pitches up and down, or rolls left and right quite aggressively.

Then along those same lines, reaching the VTOL_LAND waypoint causes the vehicle to jerk as it looks like it switches controller or something.

Any parameters we're getting wrong?  Or is it just like this for everybody in which case it's probably a bug?

igal74, 2018-10-17:
hi all,
we also have a vtol drone with the exact same issue,
and i also want it fixed,
can we help by supplying some data fro you to work with?
or even - if someone can point me in the code where the issue is, i can fix and test it on my drone, and than upload the fix

duran9dv, 2019-04-10:
Also trying to smooth out our back transitions, and the decel param does not seem to make any effect. @tridge maybe you can shed some light ? 

We're also having strong pitch up and gain in altitude, we have a lot more room to play with on these back transitions, just looking for the right params to adjust and test

IamPete1, 2021-09-07:
Lots of changes to this logic on 4.1, so I'm going to close this issue. Happy to have a look at logs and try and suggest some param changes over on the forum, if there is still problems we can open a new issue.

closed 2021-09-07 by IamPete1