ArduPilot/ardupilot #8109: ChibiOS: FMUv4 does not support DSM input

WickedShell, 2018-04-07:
#### Issue details
ChibiOS builds do not control the regulator for Spektrum receivers. (I don't see any parser hooked to it either, but without the regulator being enabled this is a moot point). RC receiver binding is also unsupported.

I started looking into adding this stuff, but it wasn't obvious how ChibiOS wants to manage board specific features. (IE do I just check if the hardware pins are defined for SPEKTRUM stuff and if so include the code, or is there a build type macro that should be checked instead?)

#### Version
Master (9e77b0cf1f13 at time of writing)

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Hardware type
Pixracer

davidbuzz, 2018-04-24:
I'd say this list is short enough, and the issues obscure enough that we can just put these into the release document as known incomplete, and ship chibios without them.

DavidIngraham, 2018-05-15:
A few others I've found while bringing up the F405-Wing:
- OSD Support (This would be an enhancement, but we have it in the F4Light HAL)
- SPI SD Card Support (This is common on lower end boards)
- Detection of all external i2c compasses

Naterater, 2018-07-12:
As of the last dev call, it was mentioned that shutter feedback was not yet active on ChibiOS. Has that been addressed?

WickedShell, 2018-07-12:
@Naterater #8886 would resolve it

tridge, 2018-07-18:
@DavidIngraham we have now implemented osd support, spi sd card support and external compasses on all boards

hdtechk, 2018-07-29:
Support for OreoLED is not on list and not working

EShamaev, 2018-08-26:
https://github.com/ArduPilot/ardupilot/pull/9285

<=FMUv3 DSM receiver binding

Gregmaan, 2018-09-14:
Hi i am using a Pixhawk 1 and would like to use the ADC inputs (RSSI, Airspeed). Are there any plans to support these inputs.
Or should I buy another board and i2c airspeed sensor?
Regards
Gregor

palm369, 2018-10-17:
Just tested 3.9.2 and RPM input on aux pin is not working. But it is checked as done here. What is the status?

tridge, 2018-10-29:
@palm369 my guess is you have not set BRD_PWM_COUNT correctly (its a common mistake)

davidbuzz, 2018-10-29:
see also here: https://github.com/ArduPilot/ardupilot/pull/9651

palm369, 2018-11-06:
@tridge RPM is only working with the NuttX firmware version, but not working on ChibiOS. I just double checked it with AP3.9.3 B1.

peterbarker, 2018-11-07:
On Tue, 6 Nov 2018, David Kaden wrote:

> @tridge RPM is only working with the NuttX firmware version, but not working
> on ChibiOS. I just double checked it with AP3.9.3 B1.

Which board and pin?

tridge, 2018-11-07:
@palm369 the most likely reason is you have BRD_PWM_COUNT setup so that the pin is assigned to PWM, not GPIO. On PX4 the conflict wasn't detected, but with ChibiOS we enforce this

palm369, 2018-11-07:
Cube, AUX5, BRD_PWM_COUNT=4 in both cases

peterbarker, 2018-11-07:
On Tue, 6 Nov 2018, David Kaden wrote:

> Cube, AUX5, BRD_PWM_COUNT=4 in both cases

Hmmm.

Could you put a copy of your parameters (or a dataflash log) up somewhere, 
please?

palm369, 2018-11-07:
@peterbarker https://www.dropbox.com/s/qfjp4mijeqqb9w2/latest%20params.param?dl=0

WickedShell, 2018-11-07:
@tridge ChibiOS isn't setting the `ardupilot_git_hash` key in the apj, and is leaving the git field from px4 as not filled out. Also depends upon `PX4Firmware/Tools/px_mkfw.py` which would need to be replaced before PX4Firmware could be removed.

peterbarker, 2018-11-07:
On Wed, 7 Nov 2018, David Kaden wrote:

> @peterbarker https://www.dropbox.com/s/qfjp4mijeqqb9w2/latest%20params.param?dl=0

Please try setting RELAY_PIN to -1

Pedals2Paddles, 2018-11-08:
Can the Solo's LED driver be ported over before the Nuttx builds are killed?

proficnc, 2018-11-08:
Whoops sorry

palm369, 2018-11-09:
@peterbarker > Please try setting RELAY_PIN to -1

That solved it, thanks. But I have not ever touched this param, it must been set automatically while upgrading to chibiOS. Shouldn't be that way IMHO

Pedals2Paddles, 2018-11-12:
Added https://github.com/ArduPilot/ardupilot/issues/9763 for the simulated DSM not working on the new IO.

Pedals2Paddles, 2018-11-16:
DSM fixed in https://github.com/ArduPilot/ardupilot/pull/9786

bchristal, 2018-11-30:
Thought I'd add to the list. Hot shoe events are not recognized by 3.9.3 Chibios. Hot shoe events work fine in 3.9.3 standard.

Naterater, 2018-11-30:
Check your BRD_PWM_COUNT and make sure that your aux channels aren’t PWM. NuttX didn’t enforce that parameter correctly. I have both trig and Cam events running master.

bchristal, 2018-12-01:
Hi Nathan,

Thanks. I think you're right, but I'm not clear on using BRD_PWM_COUNT. The default setting is 4. I assumed AUX 1-4 were PWM, AUX 5-6 General Purpose? Moving the hot shoe cable to AUX 6 (PIN 55) works. So I set BRD_PWM_COUNT to 3 (assuming AUX 1-3 are PWM and 4-6 are General Purpose) and connected the the Hot Shoe cable to Pin 53, AUX4 (Fast capture), but it is not recognized.

Do you know how BRD_PWM_COUNT affects the AUX pins?

Brian

Naterater, 2019-01-17:
@WickedShell, I see 3 items on the list that are related to the missing features that haven't been completed yet:

- [ ] aerofc-v1 support
- [ ] VRBrain support
- [ ] APJ does not include the appropriate git hash entries

Is that right? I think the wiki is a separate issue not related to missing features, and the items that are decidedly unsupported shouldn't hold up closing this issue. Maybe those could be separated into separate issues?

TimothyGold, 2019-06-17:
I tried ChiBiOS with the latest ArduCopter on PixRacer yesterday and found that Analog RSSI is not implemented.  Everything else seems to work as far as I could tell.

Zadrob, 2019-07-31:
ChibiOS SRXL protocol support for new Spektrum receivers.
They have just changed CRC. (from AR7700 to AR9030)

kd0aij, 2019-09-30:
It looks like we lost this functionality when moving to ChibiOS: ArduPilot/PX4Firmware#38

closed 2019-12-18 by tridge