ArduPilot/ardupilot #8157: ArduCopter lose propeller protection

marstears, 2018-04-13:
#### Issue details
I have a 960mm hex copter with motor thrust and power redundancy. I made one motor stop in auto mode. It was flying normally and perfectly.  But when I changed flight mode to RTL, the copter started to self-rotate and uncontrollable in YAW direction. I did this test three times and I found that when the motor stop the copter could control itselfe. But when I changed flight mode, no mater which mode I changed, it started to self-rotate. I hope the lose propeller protection could be improve.

#### Version
ArduCopter 3.5.4 3.5.5

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
960mm hex copter.
#### Hardware type
Pixhawk 2

#### Logs

rmackay9, 2018-04-23:
I don't think the motor should ever stop if the ESCs are calibrated correctly.

If it is a matter of the ESCs slowly down so quickly that the propellers come off because they are self tightening.. it may help to reduce the ATC_ACCEL_R/P/Y_MAX parameters to reduce the aggressiveness of the attitude control.  These parameters are also set during an autotune.

lthall, 2018-04-24:
Hex copters do not guarantee yaw control when one motor is lost. This is a fundamental physics based limitation.

You can maintain yaw control if your CG is not perfectly centred and you lose the right motor but any disturbance can start  the aircraft rotating in yaw.

closed 2018-04-24 by lthall