ArduPilot/ardupilot #8159: RangeFinder_VL53L0X stop working

shaddow501, 2018-04-13:
Hello

I have encountered an issue when the RangeFinder VL53L0X is connected together with Toshiba LED.

it seems that when the Toshiba LEDis connected the VL53L0X hangs / freezes and stop working.

From my investigation it is not a hardware issue but a software issue.

Things I did to investigate:
First we know that the AP_Notify Toshiba_LED_I2C is looking for "ack" from the 0x55 address to find out if the LED is exist.

1. I powered on the PixRacer without the Toshiba LED while the GPS and VL53L0X was connected and it ran without any failures for couple of hours.
2. Powered on the PixRacer with the Toshiba LED + GPS + VL53L0X and the VL53L0X stopped working after a few minutes (more or less 10 min it worked till it stop getting data - seen by "sensorrange" in "Mission planner"

Then I thought that It might be a hardware issue, And to verify if it is hardware or code issue I turned on the PixRacer with the Toshiba LED waited till it recognized it (by flashing the LEDs) and then disconnected it, this was to emulate that the code will continue to send I2C data to the Toshiba LED.

The VL53L0X stopped working a few minutes later!

So I am now sure that something with the Toshiba LED driver/code cause interference with the VL53L0X.

rmackay9, 2018-04-26:
I've added a wiki to-do issue to improve the wiki page for this lidar: https://github.com/ArduPilot/ardupilot_wiki/issues/1216

auturgy, 2018-07-30:
Does https://github.com/ArduPilot/ardupilot/pull/9024 fix this?

rmackay9, 2018-07-30:
@auturgy, yes definitely.  Let's close this issue and we can re-open if we find the next release candidate still doesn't work.  The next release candidate will be released tomorrow or the day after and should be called Copter-3.6.0-rc7

closed 2018-07-30 by rmackay9