ArduPilot/ardupilot #8187: Plane: motors may unexpectedly power-up when flight controller rebooted

rmackay9, 2018-04-18:
I am not an experienced plane user but I had the main motor on my nano talon fire up as I rebooted the flight controller.  This is the sequence of events:

- vehicle was in Manual mode and I'm pretty sure it was disarmed (otherwise the motor would have been spinning when I disarmed it I think)
- transmitter throttle was at mid (it is a transmitter with a sprung throttle)
- rebooted the flight controller
- motor spun up (luckiliy the props were off)

juvinski, 2018-04-18:
How much is the value of your esc mid pwm and how much is your controller sending esc ?

juvinski, 2018-04-18:
I had similar behavior and was the pwm neutral - the radio was sending 1455 instead the 1500 configured on the esc and every boot the motor was spinning - first I tried to decrease rc3_trim to 1455 and after to 1555 to match the correct value

rmackay9, 2018-04-18:
@juvinski, The ESC's range is about 1000 ~ 2000 pwm (i think) with 1000 being off.  My transmitter was sending in mid throttle (about 1500pwm) so the issue is likely that as the main CPU on the Pixhawk is rebooted the I/O boards starts passing through the pilot's throttle to the ESC.  My feeling is that if we receive a reboot command (via mavlink), if the vehicle is disarmed then before rebooting we should put the I/O board in a state that the motors will not spin-up.

I don't know plane well though so this is just my feeling.

Naterater, 2018-04-18:
1st question: What is your RC3_Trim set to? Is it mid, or is it = RC3_Min?

2nd question: Should we be recommending that users with a "sprung" throttle use only half of the throttle range? (AKA, RC3_Min = 1500) Sprung throttles actually work well for reverse-thrust planes.

WickedShell, 2018-04-26:
Duplicate of #7005

closed 2018-04-26 by WickedShell