ArduPilot/ardupilot #8198: Plane: send channel 9 - 12 motor data to GCS

expntly, 2018-04-20:
#### Issue details
According to what I'm seeing in MP and MavProxy, for an OctaQuadPlane X8, trying to get data for ch9out, ch10out, ch11out, and ch12out doesn't "work".  In MP it doesn't graph anything.  In MavProxy I don't see anything past chan8_raw -- i.e. there's no chan9_raw, chan10_raw, chan11_raw or chan12_raw in msg when type == RC_CHANNELS_RAW.

#### Version
AP3.8.5beta1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
OctaQuadPlane

#### Hardware type
PH2.1

WickedShell, 2018-04-20:
We do send channels 9-16, however you must be using MAVLink2 for those channels to be sent.

closed 2018-04-20 by WickedShell

expntly, 2018-04-20:
@WickedShell  Oops thanks for the note on SERVO vs RC.

I enabled MAVLink2 and this does indeed enable the other channels, however they show `0`.  Are you sure this works?  Or maybe SITL doesn't support OctaQuadPlane? I use `mavproxy -f quadplane` and then `Q_FRAME_CLASS=4`.