ArduPilot/ardupilot #8211: Preparing to add external SPI IMU for better flight

zarathustr, 2018-04-22:
Hi guys, we have released a new external IMU consisting of ADI gyros and accelerometers. The flight experience has been improved. Also the control performance is better now. We are going to upload the source codes to APM for further support. As is known, the Pixhawk, Pixhawk 3 Pro and Pixhawk 4 all have onboard external SPI connector. The IMU is then applied to these platforms with sampling frequency of 1000Hz. 

The following is a typical sampling via the PX4 with 3 ADXRS620 gyros
[image]

With the onboard MPU6000 and external IMU, two test flights have been conducted. For the first one, the MPU6000 is used for computing attitude. The control performances are shown as follows:

[image]

[image]

[image]

[image]

**While for the external IMU, the performances are:**

[image]

[image]

[image]

[image]

Note that here during the two flights, the parameters are default and the quadrotor is operated at the same position so the comparison is fair. It is obvious that the attitude rates control using ADI IMU is better which leads to better controlling behaviour on the roll and pitch. The basic version containing 3 ADXRS620 gyros and two ADXL203 accelerometers will be priced at 1999 RMB (317.7707 USD) while for the medium version it would be 2999 RMB (476.7355 USD) which replaces the yaw gyro with ADXRS642. **All the sensors have been factory calibrated** including factors, misalignment, biases and temperature drift. We hope that this is going to improve the flight experience of APM in the future!

PS: The test PX4 ulog files can be downloaded on: [website](https://github.com/zarathustr/ext_imu_test)

geofrancis, 2018-04-22:
very cool, those gyros are nice but well outside my price range, but it got me thinking does arducopter support any of the hobby grade gyros externally on the spi port? trying to vibration isolate a pixhawk is very difficult when you have lots of wires attached to it but being able to soft mount a external gyro somewhere on the airframe would do a lot of good.

rmackay9, 2018-04-23:
@zarathustr, Ok,very interesting.  looking forward to the pull request.  Please consider doing a blog post as well on ardupilot.org.

@geofrancis, ardupilot supports quite a wide range of IMUs but normally I think we only look on the internal SPI port so some code changes would be required to make it look on the external port.  The code changes would be quite small though I think.

geofrancis, 2018-04-23:
@rmackay9 I was thinking being able to mount an external IMU right in the CG could also be used to compensate for builds where the pixhawk is mounted far away from the CG.

auturgy, 2018-10-14:
@zarathustr have you progressed this?

Kiwa21, 2019-04-29:
@zarathustr Any updates ?

trollp, 2019-09-20:
Is there any further information about the implementation?

frontw, 2019-10-04:
> Is there any further information about the implementation?

+1

rmackay9, 2020-01-11:
I might close this issue because it's really more of a notification instead of an enhancement or bug report and there is a similar enhancement request to add support for high-end (aka expensive) IMUs or even full AHRSs here: https://github.com/ArduPilot/ardupilot/issues/11479.

By the way, there are some new IMUs from Invensense (and perhaps other manufacturers as well) that have a higher range (30G instead of the 16G of older IMUs) and these should improve the vehicle's resistance to high vibrations.  These IMUs are starting to appear on new flight controllers like the Holybro Durandal and the Hex CubeOrange.

closed 2020-01-11 by rmackay9