ArduPilot/ardupilot #8220: issue of Airspeed SDP3X

liang-tang, 2018-04-24:
#### Issue details
I tested SPD33 airspeed yesterday, Flight was well. But SDP33 airspeed seems weird at low airspeed, Bounce between 1.85 and 0 and discrete. Today I tested it again,  got the same result, but when i comment the `AP_Airspeed_SDP3X::_correct_pressure(float press)`, result is successive. So I geuss something wrong with `_correct_pressure`. 

#### Version
3.8.5beta2 release

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ * ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

liang-tang, 2018-04-24:
Flight test yesterday, ARSP is MS4535DO, ASP2 is SDP33 from drotek.
[image]

Comment VS not Comment (_correct_pressure)

[image]
[image]

WickedShell, 2018-04-24:
@liang-tang we need a raw log. There are a couple of known issues with this sensor and driver at this point that we are looking into. (Dataflash logs would be helpful for reviewing changes).

liang-tang, 2018-04-24:
@WickedShell Test log of the last  two pictures.
https://drive.google.com/open?id=16dqQGCcCDowIrtNSARsal4suJcGBozhg
https://drive.google.com/open?id=1_BD4MoOvPqnDPH5DTc54dYZOxrqYsk52

liang-tang, 2018-04-24:
change the follow line 
https://github.com/ArduPilot/ardupilot/blob/29e8866ea910b6b5ce31f416ec048b6ba1a6b53e/libraries/AP_Airspeed/AP_Airspeed_SDP3X.cpp#L255
to `float dp_pitot = 28557670.0f * (1.0f - 1.0f / (1.0f + (float)powf((flow_SDP3X / 5027611.0f), 1.227924f)));`
 can solve this  problem. I dont konw why , but it works actually.

liang-tang, 2018-04-25:
PR 

> https://github.com/ArduPilot/ardupilot/pull/8229

Test log

> https://drive.google.com/open?id=14wWUNQ7xnyFeHHD8vgJphzxa3UIQTxIv

[image]

Naterater, 2018-12-17:
I've had a lot of good flights with this sensor, and seeing how the PR was merged, I'm going to close this issue.

closed 2018-12-17 by Naterater