ArduPilot/ardupilot #8222: Differentiate RTL altitude from Failsafe (RTL) altitude

vag01, 2018-04-24:
#### Desired feature

It would be very helpful to have 2 different altitudes for RTL:

1. RTL Altitude: This is when the RTL is requested by the pilot, by switching to RTL mode.
In most of the cases the current aircraft’s altitude can be used and in anyway the pilot has the control and can climb if needed. This can be for instance the actual RTL_ALT.

2. RTL Failsafe Altitude: This happens when the RTL is activated by the failsafe (if for instance the aircraft is out of the RC’s range) 
A second parameter, that could be RTL_FS_ALT for instance, could be implemented to reach an altitude before returning to launch in order avoid obstacles as the aircraft will be flying without any control from the pilot.

Thank you

#### Version
3.5.5

#### Platform
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

JeyPi, 2018-06-17:
+1 - is somebody considering this?
My today's FPV experience:
I took off from a spot 280 meters AMSL (altitude above mean sea level)
I was remotely landing the plane while in the stabilized mode at an RC field that was a 1000 meters away from me and a bit lower - 240 AMSL.
I was gliding in for landing. All was great. Just before touch down, the plane lost RC link and the RTL mode kicked in. I was no longer in control. The ardupilot software took a right turn towards the Take off position without any climb. Just a turn. That of course was followed by a crush as the plane was so low and the tip of the wing touched the ground. My RTL altitude was set to 100 meters.
I suggest the code should be adjusted to climb the plane to a safe altitude before turning to the home position. I understand this could not suite to all platforms so at least this behavior could be user pick able.
The only data that survived was the camera footage. The flight deck data card was destroyed.

[Video of the event here](https://drive.google.com/open?id=1M_ZMg-GumfTnKhr5usVDDcZS5AaffCR2)

As you can see, the only thing needed was a little climb before the turn and plane would had been saved.

iforce2d, 2018-09-27:
@JeyPi  Wow really bad luck... but was it in CIRCLE or RTH mode there? In my experience RTH will start to climb immediately and even and quite violently so in some cases, it's unmistakable. The default is to CIRCLE (short failsafe) ie. bank to the right for a few seconds with no attempt to climb, before switching into RTH proper. It kinda looks like that's what happened in your case.
For a situation where you'll be landing at a different location to the launch point, you could also look into rally points for RTL action. http://ardupilot.org/copter/docs/common-rally-points.html
Disabling the short failsafe and setting a rally point upwind of the runway to be landed on might prevent a wing-tip catching on the ground.

Redshifft, 2018-12-15:
I would like to push for a CLIMB first option for Arduplane, very surprised this has not been pushed for before as it it the logical  failsafe action for a Plane - Climb clear of obstacles or ground before turning hard for home.
@rmackay9  should I start this as a separate Plane request?

rmackay9, 2018-12-17:
@Redshifft,

probably best to raise a new enhancement request.  This issue was raised for Copter and the failsafe behaviour is mostly different for each vehicle so with a new enhancement request it's less likely to be lost.

Redshifft, 2018-12-17:
Ok, Thanks