ArduPilot/ardupilot #8226: Copter: "Delay Until UTC" function doesn't gracefully handle times in the near past

gdivr4, 2018-04-24:
#### Issue details
"Delay Until UTC Time" functionality (do_nav_delay) is fragile. If I say delay until 12:00:00 and my platform doesn't make it to that step in the mission until 12:00:01, the nav_delay_time_max variable will be set to 23.99 hours. This limits the usefulness of this command, as time delays need to be large enough that you couldn't possibly miss them due to wind, etc. 

#### Version
Arducopter 3.5

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Sim

#### Hardware type
Sim

#### Logs
N/A

peterbarker, 2018-05-10:
So this is currently working as-designed.

I don't think the original intent of `MAV_CMD_NAV_DELAY` was to do tight timing.  It was more for "stay on the ground until this time every day, then fly your mission, then land again, and then we'll do the same thing tomorrow"

We need to do better here.  The question is - how?

We could use parameter five as an absolute time of day, perhaps a UTC time in seconds (delta from midnight 2018-01-01?).  This would become a "NOT_BEFORE" field - so we delay at this mission item until that UTC time has passed (or we're at that UTC time).  This is significantly different to the first three fields; being an absolute time it becomes invalid once that time has passed, whereas both other modes ('delay for n seconds' and 'delay until this hour,  minute, second') are valid any time you boot the vehicle.

Another option is to use parameter five as a bitfield.   Bit 0 could mean, "if we're within 1 minute of the time specified in parameters 2,3 and 4 then consider the item verified".  That seems kinda awful to me.

gdivr4, 2018-05-15:
I think a UTC time would ultimately be the least ambiguous option, but that would obviously require some changes on the ground control side to make the hooks available to the user. 

A less invasive approach would be to cap the maximum wait time, changing from a 0-24 hour window to a -12 to +12 hour time window, where anything greater than 12 hours would "overflow" and be considered a past time. You could make the max time an EEPROM param to allow for reverse compatibility (set to 24 hours to get current behavior).