ArduPilot/ardupilot #8237: optical flow does not work

mohand150, 2018-04-25:
#### Issue details
I'm trying to run the code on ocpoc zynq mini
I use the code ardupilot
I activate the optical flow in: `AP_GROUPINFO ("_ ENABLE", 0, OpticalFlow, _enabled, 1),`
I add a small display in the opticalflow code
[image]

then copy the executable in the ocpoc but when I run it I can not find the sensor
there is something else to touch in the code
optical flow does not work
[image]
can someone  help me please
#### Version
_What version was the issue encountered with_

#### Platform
[  ] Copter
#### Hardware type
OcPoc

mohand150, 2018-04-25:
@tridge there is something else to change in the code ?

DavidIngraham, 2018-04-26:
Hello @mohand150,
I have updated the title to more precisely reflect the issue you are experiencing. Could you provide more information about the problem you are experiencing?
Specifically:
Are you running master?
Have you followed the instructions in the Wiki? 
Has the PX4Flow firmware been updated?

mohand150, 2018-04-27:
for hard you can see that I used an OcPoc in which I run the Ardupilot Master Firmware I downloaded
[image]

I configured the PX4flow as indicated in the wiki
in parity TESTING THE SENSOR `http://ardupilot.org/copter/docs/common-px4flow-overview.html#testing-the-sensor`

they say that normally when I connect the Pxhaw (in my case OcPoc) normally I will get opt_m_x, opt_m_y and an opt_qua values, and distance_ground.
I change the code a bit as you can see in the attached file so that it shows me in the console But I can not get the data
I could have the sensor bus as you see in the following picture,

[image]

I have 2 problems ???
1 do not get the valid ground clearance (always 444 or 454, whatever the distance) ?
2 the update function does not work ?

auturgy, 2018-04-27:
You need to have a lidar or other rangefinder attached and set up: the sonar on the flow sensor is not used.

mohand150, 2018-04-27:
hello @auturgy
I already bought a lidar lite V3
I'm trying to run the PX4 first in XY horizontal control and then I'll add the Lidar for the vertical control
so how can i use it directly ???
is it just a modfier parameter in QGS or should i change the ardupilot code ???

auturgy, 2018-04-27:
Ah! QGC.
I just rewrote the wiki to help with qgc setup. Not merged yet.
https://github.com/ArduPilot/ardupilot_wiki/pull/1221 might help.

Regards,

James

mohand150, 2018-04-27:
as you described on the wiki
I followed all the steps, I use mission planner as you
1-Upgrade the PX4Flow sensor's firmware
2-Focusing the lens
3- Connect to the Pixhawk (in my case OcPoc)
4-Enabling the sensor (in the code of Px4Flow.cpp) and in mission palnner
5-but here; Testing the sensor http://ardupilot.org/copter/docs/common-px4flow-overview.html#testing-the-sensor

I dont have data Opticalflow
is this a relation with the flight controller (OcPoc)
regards,
 Alex

mohand150, 2018-04-27:
when I connect Px4flow Directly to the PC I get the data in Mission planner Like this

[image]

but when I use OcPoc as flight controller I detect nothing in APM planner2 
how can i get the sensor data ?

mohand150, 2018-04-27:
thank you james 
I have already tried without midifying the code but it does not work but as it does not detect the sensor so I had to fix the address of the sensor, now I can detect it but these data are false
(he does not update the data) and he does not display them in APM planner

magicrub, 2018-04-27:
that wiki page update https://github.com/ArduPilot/ardupilot_wiki/pull/1221 has been merged

auturgy, 2018-05-15:
Closing, as this seems to be a support issue, not a confirmed bug, at this stage.
A request for assistance in the forum (discuss.ardupilot.org) will reach a wider audience than here, and in this case I think asking Aerotenna directly is the best course of action.  If an issue in the code is identified, please raise a new issue and link to this one.

closed 2018-05-15 by auturgy