ArduPilot/ardupilot #8238: Compass NXP MAG3110 not working

ukmook, 2018-04-25:
Compass NXP MAG3110 not working, have tested with sparky2 and revo, have tested with the same scenario HMC5483 and that works.
MissionPlanner Alarm is not present but the magfield in status tab shows zero
Tested the mag in INAV and it works

#### Version
Built from master chibios

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Tested on bench

#### Hardware type
Sparky2, Revo & qUark
GPS board with MAG3110

#### Logs
Dataflash not working with either board

night-ghost, 2018-05-08:
Should be fixed now

ukmook, 2018-05-09:
Thankyou, i am getting values in status window Mission Planner but not a valid heading. Also when doing a calibration the field populates it self without moving the vehicle, it constantly does this (bar gets to end and starts again) until it hangs. Tested against a QMC5883L which works perfectly on same bus.

night-ghost, 2018-05-10:
> i am getting values in status window Mission Planner but not a valid heading

for external compass you should to set correct orientation

>Also when doing a calibration the field populates it self without moving the vehicle

more details please